pybamm-team/PyBaMM #2461: [Bug]: cyclic_ageing.py example does not work as is

skywo1f, 2022-11-16:
### PyBaMM Version

>>> pybamm.version() (21, 10)

### Python Version

3.8

### Describe the bug

python3 cyclic_ageing.py gives the error:

FileNotFoundError: Could not find parameter Mohtat2020

The line :

param = pb.ParameterValues("Mohtat2020")

needs to be replaced with:

chemistry = pb.parameter_sets.Mohtat2020
param = pb.ParameterValues(chemistry=chemistry)

### Steps to Reproduce

python3 cyclic_ageing.py

### Relevant log output

```shell
Also it would be nice if the cyclic_ageing simulation showed the drop in battery capacity over the cycles.
```

tinosulzer, 2022-11-16:
Why are you using version 21.10 ? You should update to the latest version (22.10)

Here is an example with capacity vs cycle number https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/simulating-long-experiments.ipynb

skywo1f, 2022-11-17:
thanks!

It complained a fair amount, but it did finish:
`/batterySims/PyBaMM/examples/scripts$ python3 cycling_ageing.py 
2022-11-17 07:10:34.093 - [NOTICE] callbacks.on_cycle_start(174): Cycle 1/6 (10.163 ms elapsed) --------------------
2022-11-17 07:10:34.093 - [NOTICE] callbacks.on_step_start(182): Cycle 1/6, step 1/5: Charge at 1 C until 4.2 V
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
2022-11-17 07:10:34.990 - [NOTICE] callbacks.on_step_start(182): Cycle 1/6, step 2/5: Hold at 4.2 V until C/10
2022-11-17 07:10:35.276 - [NOTICE] callbacks.on_step_start(182): Cycle 1/6, step 3/5: Rest for 5 minutes
2022-11-17 07:10:35.457 - [NOTICE] callbacks.on_step_start(182): Cycle 1/6, step 4/5: Discharge at 1 C until 2.8 V
At t = 0.00640064, , mxstep steps taken before reaching tout.
At t = 0.00265859 and h = 1.45678e-30, the corrector convergence failed repeatedly or with |h| = hmin.
2022-11-17 07:10:39.706 - [NOTICE] callbacks.on_step_start(182): Cycle 1/6, step 5/5: Rest for 5 minutes
2022-11-17 07:10:39.920 - [NOTICE] callbacks.on_cycle_start(174): Cycle 2/6 (5.837 s elapsed) --------------------
2022-11-17 07:10:39.920 - [NOTICE] callbacks.on_step_start(182): Cycle 2/6, step 1/5: Charge at 1 C until 4.2 V
2022-11-17 07:10:40.401 - [NOTICE] callbacks.on_step_start(182): Cycle 2/6, step 2/5: Hold at 4.2 V until C/10
2022-11-17 07:10:40.565 - [NOTICE] callbacks.on_step_start(182): Cycle 2/6, step 3/5: Rest for 5 minutes
2022-11-17 07:10:40.622 - [NOTICE] callbacks.on_step_start(182): Cycle 2/6, step 4/5: Discharge at 1 C until 2.8 V
At t = 0.00653781, , mxstep steps taken before reaching tout.
At t = 0.00279576, , mxstep steps taken before reaching tout.
2022-11-17 07:10:43.217 - [NOTICE] callbacks.on_step_start(182): Cycle 2/6, step 5/5: Rest for 5 minutes
2022-11-17 07:10:43.348 - [NOTICE] callbacks.on_cycle_start(174): Cycle 3/6 (9.265 s elapsed) --------------------
2022-11-17 07:10:43.348 - [NOTICE] callbacks.on_step_start(182): Cycle 3/6, step 1/5: Charge at 1 C until 4.2 V
2022-11-17 07:10:43.799 - [NOTICE] callbacks.on_step_start(182): Cycle 3/6, step 2/5: Hold at 4.2 V until C/20
2022-11-17 07:10:44.113 - [NOTICE] callbacks.on_step_start(182): Cycle 3/6, step 3/5: Rest for 30 minutes
2022-11-17 07:10:44.431 - [NOTICE] callbacks.on_step_start(182): Cycle 3/6, step 4/5: Discharge at C/3 until 2.8 V
2022-11-17 07:10:45.721 - [NOTICE] callbacks.on_step_start(182): Cycle 3/6, step 5/5: Rest for 30 minutes
2022-11-17 07:10:46.095 - [NOTICE] callbacks.on_cycle_start(174): Cycle 4/6 (12.013 s elapsed) --------------------
2022-11-17 07:10:46.095 - [NOTICE] callbacks.on_step_start(182): Cycle 4/6, step 1/5: Charge at 1 C until 4.2 V
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
2022-11-17 07:10:46.676 - [NOTICE] callbacks.on_step_start(182): Cycle 4/6, step 2/5: Hold at 4.2 V until C/20
2022-11-17 07:10:46.868 - [NOTICE] callbacks.on_step_start(182): Cycle 4/6, step 3/5: Rest for 30 minutes
2022-11-17 07:10:47.075 - [NOTICE] callbacks.on_step_start(182): Cycle 4/6, step 4/5: Discharge at 1 C until 2.8 V
At t = 0.00670047 and h = 1.62313e-77, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 0.00295842, , mxstep steps taken before reaching tout.
2022-11-17 07:10:49.150 - [NOTICE] callbacks.on_step_start(182): Cycle 4/6, step 5/5: Rest for 30 minutes
2022-11-17 07:10:49.429 - [NOTICE] callbacks.on_cycle_start(174): Cycle 5/6 (15.346 s elapsed) --------------------
2022-11-17 07:10:49.429 - [NOTICE] callbacks.on_step_start(182): Cycle 5/6, step 1/5: Charge at 1 C until 4.2 V
2022-11-17 07:10:49.851 - [NOTICE] callbacks.on_step_start(182): Cycle 5/6, step 2/5: Hold at 4.2 V until C/20
2022-11-17 07:10:50.039 - [NOTICE] callbacks.on_step_start(182): Cycle 5/6, step 3/5: Rest for 30 minutes
2022-11-17 07:10:50.247 - [NOTICE] callbacks.on_step_start(182): Cycle 5/6, step 4/5: Discharge at 2 C until 2.8 V
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:849: Calculating Jacobian failed
At t = 0.00255544 and h = 9.63594e-36, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 0.00255544, , mxstep steps taken before reaching tout.
2022-11-17 07:10:52.187 - [NOTICE] callbacks.on_step_start(182): Cycle 5/6, step 5/5: Rest for 30 minutes
2022-11-17 07:10:52.472 - [NOTICE] callbacks.on_cycle_start(174): Cycle 6/6 (18.389 s elapsed) --------------------
2022-11-17 07:10:52.472 - [NOTICE] callbacks.on_step_start(182): Cycle 6/6, step 1/5: Charge at 1 C until 4.2 V
2022-11-17 07:10:52.863 - [NOTICE] callbacks.on_step_start(182): Cycle 6/6, step 2/5: Hold at 4.2 V until C/20
2022-11-17 07:10:53.062 - [NOTICE] callbacks.on_step_start(182): Cycle 6/6, step 3/5: Rest for 30 minutes
2022-11-17 07:10:53.271 - [NOTICE] callbacks.on_step_start(182): Cycle 6/6, step 4/5: Discharge at 3 C until 2.8 V
At t = 0.00371278 and h = 3.56605e-30, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 0.00371278 and h = 5.98417e-20, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 0.00121808, , mxstep steps taken before reaching tout.
At t = 0.00121808 and h = 3.2765e-31, the corrector convergence failed repeatedly or with |h| = hmin.
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:849: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:849: Calculating Jacobian failed
At t = 0.000594407 and h = 4.56764e-51, the corrector convergence failed repeatedly or with |h| = hmin.
2022-11-17 07:10:56.527 - [NOTICE] callbacks.on_step_start(182): Cycle 6/6, step 5/5: Rest for 30 minutes
2022-11-17 07:10:56.817 - [NOTICE] callbacks.on_experiment_end(222): Finish experiment simulation, took 22.734 s
`

closed 2022-11-17 by tinosulzer