pybamm-team/PyBaMM #2490: RuntimeError while trying to run drive cycle in experiment with initial SOC

Mrzhang-hub, 2022-11-26:
### Discussed in https://github.com/pybamm-team/PyBaMM/discussions/2489

<div type='discussions-op-text'>

<sup>Originally posted by **Mrzhang-hub** November 26, 2022</sup>
Hi! I am trying to run a drive cycle in an experiment but it keeps failing. Part of the code as follows:

```
options = {"lithium plating":"irreversible","thermal": "lumped"}
model = pybamm.lithium_ion.SPMe(options=options)
param = pybamm.ParameterValues('OKane2022')
soc_init = 0.2
drive_cycle_current = pd.read_csv("StandardCycle/UDDS.csv", comment="#", header=None).to_numpy()
exp = pybamm.Experiment([("Run UDDS_A (A)")],drive_cycles={"UDDS_A": drive_cycle_current})
sim = pybamm.Simulation(model=model, parameter_values=param, experiment=exp)
sol = sim.solve(initial_soc=soc_init)

```

It gives the error:

```
RuntimeError: .../casadi/core/slice.cpp:83: Assertion "stop<=len" failed:
Slice (start=0, stop=20, step=1) out of bounds with supplied length of 1
```

But when I remove the lithium plating option, the code works well.  

Why? Can you help me handle this? @rtimms  @tinosulzer </div>

tinosulzer, 2022-11-26:
I can't run your code to test this, since I don't have the file `StandardCycle/UDDS.csv`

Mrzhang-hub, 2022-11-26:
> I can't run your code to test this, since I don't have the file `StandardCycle/UDDS.csv`

This file is the same as the `UDDS.csv` in `pybamm/input/drive_cycles`

tinosulzer, 2022-11-26:
```python3
import pybamm
import numpy as np
import pandas as pd

options = {"lithium plating": "irreversible", "thermal": "lumped"}
model = pybamm.lithium_ion.SPMe(options=options)
param = pybamm.ParameterValues("OKane2022")
soc_init = 0.2
drive_cycle_current = pd.read_csv(
    "pybamm/input/drive_cycles/UDDS.csv", comment="#", header=None
).to_numpy()
exp = pybamm.Experiment(
    [("Run UDDS_A (A)")], drive_cycles={"UDDS_A": drive_cycle_current}
)
sim = pybamm.Simulation(model=model, parameter_values=param, experiment=exp)
sol = sim.solve(initial_soc=soc_init)
sol.plot()
```
runs without error

Mrzhang-hub, 2022-11-26:
> ```python
> import pybamm
> import numpy as np
> import pandas as pd
> 
> options = {"lithium plating": "irreversible", "thermal": "lumped"}
> model = pybamm.lithium_ion.SPMe(options=options)
> param = pybamm.ParameterValues("OKane2022")
> soc_init = 0.2
> drive_cycle_current = pd.read_csv(
>     "pybamm/input/drive_cycles/UDDS.csv", comment="#", header=None
> ).to_numpy()
> exp = pybamm.Experiment(
>     [("Run UDDS_A (A)")], drive_cycles={"UDDS_A": drive_cycle_current}
> )
> sim = pybamm.Simulation(model=model, parameter_values=param, experiment=exp)
> sol = sim.solve(initial_soc=soc_init)
> sol.plot()
> ```
> 
> runs without error

[image]
The path is the only difference between your code and mine, but it still gives this error.

tinosulzer, 2022-11-26:
What happens if you use the path I used?

Mrzhang-hub, 2022-11-26:
> What happens if you use the path I used?

Got the same error. I guess maybe it is caused by the version mismatch between pybamm and casadi ? Which version of casadi does you install?

tinosulzer, 2022-11-26:
Could be. Try on 22.10

Mrzhang-hub, 2022-11-27:
> Could be. Try on 22.10

I have upgraded pybamm into 22.10, but it still goes wrong.

tinosulzer, 2022-12-02:
We won't be able to fix this if we can't reproduce ourselves. Could you try with exactly this code on v22.10
```python3
import pybamm
import numpy as np
import pandas as pd

options = {"lithium plating": "irreversible", "thermal": "lumped"}
model = pybamm.lithium_ion.SPMe(options=options)
param = pybamm.ParameterValues("OKane2022")
soc_init = 0.2
drive_cycle_current = pd.read_csv(
    "pybamm/input/drive_cycles/UDDS.csv", comment="#", header=None
).to_numpy()
exp = pybamm.Experiment(
    [("Run UDDS_A (A)")], drive_cycles={"UDDS_A": drive_cycle_current}
)
sim = pybamm.Simulation(model=model, parameter_values=param, experiment=exp)
sol = sim.solve(initial_soc=soc_init)
sol.plot()
```
(not your `StandardCycles/UDDS.csv`)

rtimms, 2023-05-02:
@Mrzhang-hub is this still an issue? can you provide an update please?

closed 2023-05-15 by rtimms