pybamm-team/PyBaMM #2495: Remove `.diff`?

tinosulzer, 2022-11-26:
It's not used in any of the models, now that we have removed composite/first-order lead-acid models

brosaplanella, 2022-11-28:
What was it for originally?

tinosulzer, 2022-11-28:
Symbolic differentiation e.g. `(c**2).diff(c)` returns `2*c`

tinosulzer, 2022-12-02:
Yes, OCV (computing `dUdT`) was an early intended use case, before I found out everyone just gives `dUdT` explicitly in practice

tinosulzer, 2022-12-02:
In the lead-acid models, it was used to compute the first-order term of the Taylor expansion in the asymptotic expansion (e.g. `j0(c).diff(c)`). SPMe circumvents this by just using the composite term

brosaplanella, 2022-12-02:
The only case where that would be useful that I can think is in GITT, where you need U'(c). However, in practice, that's hard to calculate so you probably want to manually prescribe it.

tinosulzer, 2022-12-02:
Ok yeah, let's keep it

closed 2022-12-02 by tinosulzer