pybamm-team/PyBaMM #2498: [Bug]: ContextualVersionConflict jupyter-client

dion-w, 2022-11-28:
### PyBaMM Version

22.10.post1

### Python Version

3.8.6

### Describe the bug

Hello the bug happens, when i try to run a simulation on a HPC cluster. At first pybamm said it can't run simulations due to a parameter issue with different jupyter versions. Although i have the most up to date version of jupyter on the cluster and am using a venv.

### Steps to Reproduce

Unfortunately i do not know how i could make the environment on the cluster available to you.

### Relevant log output

```shell
Installing collected packages: wcwidth, prompt-toolkit, ptyprocess, pexpect, parso, jedi, pygments, backcall, pickleshare, traitlets, decorator, ipython, jupyter-core, tornado, pyzmq, six, python-dateutil, jupyter-client, ipykernel
Successfully installed backcall-0.2.0 decorator-5.1.1 ipykernel-6.18.1 ipython-8.7.0 jedi-0.18.2 jupyter-client-7.4.7 jupyter-core-5.1.0 parso-0.8.3 pexpect-4.8.0 pickleshare-0.7.5 prompt-toolkit-3.0.33 ptyprocess-0.7.0 pygments-2.13.0 python-dateutil-2.8.2 pyzmq-24.0.1 six-1.16.0 tornado-6.2 traitlets-5.5.0 wcwidth-0.2.5
(venv) [cg5880@uc2n996 mains_and_stuff]$ python main-WLTC-options.py
Traceback (most recent call last):
  File "main-WLTC-options.py", line 77, in <module>
    af.perform_WLTC_studies(values, options)
  File "/pfs/data5/home/kit/iam-et/cg5880/PyBaMM/mains_and_stuff/aux_functions.py", line 700, in perform_WLTC_studies
    param = pybamm.ParameterValues("Ecker2015")
  File "/pfs/data5/home/kit/iam-et/cg5880/PyBaMM/pybamm/parameters/parameter_values.py", line 81, in __init__
    if isinstance(values, str) and values in pybamm.parameter_sets:
  File "/opt/bwhpc/common/devel/python/3.8.6_gnu_10.2/lib/python3.8/_collections_abc.py", line 666, in __contains__
    self[key]
  File "/pfs/data5/home/kit/iam-et/cg5880/PyBaMM/pybamm/parameters/parameter_sets.py", line 49, in __getitem__
    return self.__load_entry_point__(key)()
  File "/pfs/data5/home/kit/iam-et/cg5880/PyBaMM/pybamm/parameters/parameter_sets.py", line 59, in __load_entry_point__
    ps = self.__all_parameter_sets[key] = ps.load()
  File "/opt/bwhpc/common/devel/python/3.8.6_gnu_10.2/python_gnu_packages/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2460, in load
    self.require(*args, **kwargs)
  File "/opt/bwhpc/common/devel/python/3.8.6_gnu_10.2/python_gnu_packages/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2483, in require
    items = working_set.resolve(reqs, env, installer, extras=self.extras)
  File "/opt/bwhpc/common/devel/python/3.8.6_gnu_10.2/python_gnu_packages/lib/python3.8/site-packages/pkg_resources/__init__.py", line 790, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (jupyter-client 6.1.7 (/pfs/data5/software_uc2/bwhpc/common/devel/python/3.8.6_gnu_10.2/python_gnu_packages/lib/python3.8/site-packages), Requirement.parse('jupyter-client>=7.0.0'), {'jupyter-console'})
```

tinosulzer, 2022-11-29:
Looks like an entry-points issue @awadell1 .

This solution might work https://github.com/click-contrib/click-plugins/issues/31#issuecomment-748349583

In the meantime, @dion-w , could you try updating your jupyter client to `>=0.7.0`

awadell1, 2022-11-29:
[entrypoints](https://github.com/takluyver/entrypoints) is in maintenance-only mode and recommends using `importlib-metadata`. This looks to be the replacement to pkg_resources so probably the move regardless. 

That said it's not immediately clear (I didn't look long) if `importlib.metadata` also requires all requirements to be available. Why is PyBamm not installing all of its requirements? Would #2457 fix this?

tinosulzer, 2022-11-30:
@dion-w can you test from the branch `issue-2498-entry-points` to see if it fixes your problem?

closed 2022-11-30 by tinosulzer

dion-w, 2022-12-03:
@tinosulzer just wanted to mention, i tested the new version on the cluster and it works like a charm. Thanks a lot!

PR #2500 (2022-11-30): #2498 try different entry points method
# Description

Fixes #2498 
See https://github.com/click-contrib/click-plugins/issues/31#issuecomment-748349583

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an i [... cut]