pybamm-team/PyBaMM #2507: Use `n_Li` or `C_Li` as the input for electrode SOH calculation?

tinosulzer, 2022-11-29:
`n_Li` is the total moles of cyclable lithium
`C_Li = n_Li * F / 3600` is the total capacity of cyclable lithium

`n_Li` is better because it's what's in the Mohtat reference paper. `C_Li` is better because it has the same units as C_n, C_p, C and so doesn't require the factor of F/3600 to convert.

I would lean towards switching to `C_Li`.

brosaplanella, 2022-12-02:
You mean as an internal Python variable or as an external "plotting" variable?

tinosulzer, 2022-12-02:
For the variable we pass in the `inputs` dictionary.
We can have both as plotting variables

brosaplanella, 2022-12-02:
Then I am happy with either of them, so if you prefer `Q_Li` I vote for it too.

tinosulzer, 2022-12-12:
Done in #2508

closed 2022-12-12 by tinosulzer