pybamm-team/PyBaMM #2515: Issue while doing cycling experiments

Aarthi2596, 2022-12-02:
### PyBaMM Version

v22.10

### Python Version

3.10

### Describe the bug

I am trying to run the basic experiment shown in Tutorial 5 with multiple cycles. When I put the paranthesis () indicating cycles, simulation is not running and I am getting operating conditions should be strings, not class tuple. Then I removed the paranthesis () inside square bracket and the simulation is running properly. But now I am not able to simulate specific cycles. For example, when I run the simulation, sim.solution.cycles, I am getting solution has no attribute cycles. Do I need to install something or missing something here? Thanks in advance

### Steps to Reproduce

experiment = pybamm.Experiment(
    [
        "Discharge at 5A until 2.5 V",
        "Rest for 1 hour",
        "Charge at 15W until 4.2 V",
        "Hold at 4.2 V until 0.01A"
    ] *3
)
sim = pybamm.Simulation(model, experiment = experiment, parameter_values = params)
sim.solve()
sim.plot(["Terminal voltage [V]", "Current [A]", "Terminal power [W]"])

sim.solution.cycles

### Relevant log output

```shell
AttributeError                            Traceback (most recent call last)
~\AppData\Local\Temp\ipykernel_46468\1236587394.py in <cell line: 1>()
----> 1 sim.solution.cycles

AttributeError: 'Solution' object has no attribute 'cycles'
```

rtimms, 2022-12-02:
Can you try this?

```python3
import pybamm

parameter_values = pybamm.ParameterValues("Chen2020")
experiment = pybamm.Experiment(
    [
        (
            "Discharge at 5A until 2.5 V",
            "Rest for 1 hour",
            "Charge at 15W until 4.2 V",
            "Hold at 4.2 V until 0.01A",
        )
    ]
    * 3
)
model = pybamm.lithium_ion.DFN()

sim = pybamm.Simulation(
    model,
    experiment=experiment,
    parameter_values=parameter_values,
)
sim.solve()
sim.solution.cycles
sim.plot()
```

Aarthi2596, 2022-12-02:
When I tried the above mentioned code, I am getting this following error:

TypeError                                 Traceback (most recent call last)
~\AppData\Local\Temp\ipykernel_46468\308321729.py in <cell line: 1>()
----> 1 experiment = pybamm.Experiment(
      2     [
      3         (
      4         "Discharge at 1C until 2.5 V",
      5         "Rest for 1 hour",

D:\Programs64\WPy64-31080\python-3.10.8.amd64\lib\site-packages\pybamm\experiments\experiment.py in __init__(self, operating_conditions, parameters, period)
     47         self.period = self.convert_time_to_seconds(period.split())
     48         self.operating_conditions_strings = operating_conditions
---> 49         self.operating_conditions, self.events = self.read_operating_conditions(
     50             operating_conditions
     51         )

D:\Programs64\WPy64-31080\python-3.10.8.amd64\lib\site-packages\pybamm\experiments\experiment.py in read_operating_conditions(self, operating_conditions)
     84                 events.append(next_event)
     85             else:
---> 86                 raise TypeError(
     87                     """Operating conditions should be strings, not {}. For example: {}
     88                     """.format(

TypeError: Operating conditions should be strings, not <class 'tuple'>. For example: 

    Discharge at 1C for 0.5 hours,
    Discharge at C/20 for 0.5 hours,
    Charge at 0.5 C for 45 minutes,
    Discharge at 1 A for 90 seconds,
    Charge at 200mA for 45 minutes (1 minute period),
    Discharge at 1 W for 0.5 hours,
    Charge at 200 mW for 45 minutes,
    Rest for 10 minutes (5 minute period),
    Hold at 1 V for 20 seconds,
    Charge at 1 C until 4.1V,
    Hold at 4.1 V until 50 mA,
    Hold at 3V until C/50,

tinosulzer, 2022-12-02:
Can you do `print(pybamm.__version__)`? It looks like you have an old version.
I'm also surprised you're using python 3.10, since pybamm doesn't with python 3.10

Aarthi2596, 2022-12-02:
I am using 0.2.4 version. Pybamm won't work with python 3.10 version?

closed 2022-12-02 by tinosulzer

Aarthi2596, 2022-12-02:
In order to update pybamm version, do I need to uninstall pybamm and install it again or do we have separate command for that? Is it okay if I still use Python 3.10, once I upgrade pybamm version?

tinosulzer, 2022-12-02:
> In order to update pybamm version, do I need to uninstall pybamm and install it again or do we have separate command for that? 

In general `pip install -U pybamm` or `conda update python` should work (depending on which one you used the first time to install)

> Is it okay if I still use Python 3.10, once I upgrade pybamm version?

No, you have to use python 3.8 or 3.9. With conda you can create a virtual environment with the appropriate python version

```bash
conda create -n pybamm-env python=3.9
conda activate pybamm-env
conda install pybamm