pybamm-team/PyBaMM #2520: Boolean operator not defined for Symbols

Ma2922, 2022-12-04:
### PyBaMM Version

22.9

### Python Version

3.9

### Describe the bug

I want to write a piecewise function , but there is an error:
Input In [22], in Ds_p(sto, T)
     18 def Ds_p(sto,T):
---> 19     if T>293.15:
     20         if sto<0.8:
     21             diff=1.0*10**(-11)*(6-5*sto)
File F:\ananconda2\lib\site-packages\pybamm\expression_tree\symbol.py:660, in Symbol.__bool__(self)
    659 def __bool__(self):
--> 660     raise NotImplementedError("Boolean operator not defined for Symbols.")

NotImplementedError: Boolean operator not defined for Symbols.

### Steps to Reproduce

def Ds_p(sto,T):
    if T>293.15:
        if sto<0.8:
            diff=1.0*10**(-11)*(6-5*sto)
        else:
            diff=2.3362*10**(-11)
    return diff
params["Positive electrode diffusivity [m2.s-1]"]=Ds_p

tinosulzer, 2022-12-04:
You can't use if-else statements when defining expressions. You have to use heavisides i.e. `(T > 293.15) * ((sto<0.8) * 1.010**(-11) * (6-5*sto) + (sto <= 0.8) * 2.3362180**(-11))` instead

Could you open a PR updating the error message to explain this?

Ma2922, 2022-12-05:
thank you very much for solving my problem.

tinosulzer, 2022-12-19:
@Ma2922 it would be nice if you could open a PR updating the error message to make it clearer, in case anyone else comes across this problem

ayeankit, 2023-03-03:
hey @tinosulzer , Is this still open? I would like to work on it.

tinosulzer, 2023-03-03:
Yes, you can work on it

ayeankit, 2023-03-03:
Hey @tinosulzer , Can you please tell more about this issue and where to implement it?

tinosulzer, 2023-03-03:
This error message: https://github.com/pybamm-team/PyBaMM/blob/14040748098322c82c8cd10998b4a57df14a00ce/pybamm/expression_tree/symbol.py#L649

needs to change to:

"Boolean operator not defined for Symbols. You might be seeing this message because you are trying to specify an if statement based on the value of a symbol, e.g. 
```python
if x < 0:
    y = 1
else:
   y = 2
```

In this case, use heaviside functions instead:
```python
y = 1 * (x < 0) + 2 * (x >= 0)
```
"

That's the end of the error message. But if you want to do more, you could also implement this function in `binary_operators.py`:

```python3
def if_else(condition, value_if_true, value_if_false):
    return value_if_true * condition + value_if_false * (1 - condition)
```

with associated tests and documentation

HarshvirSandhu, 2024-01-25:
Hello @tinosulzer 
I would like to work on this issue

arjxn-py, 2024-01-25:
Sure @HarshvirSandhu, feel free to take this up.
Assigned you for the time being

PR #3765 (2024-01-25): Fix error message for boolean operator on symbols
# Description

Update error message for Boolean operators on symbols

Fixes #2520
closed 2024-01-25 by kratman