pybamm-team/PyBaMM #2524: Google Colab fails to install PyBaMM

brosaplanella, 2022-12-05:
### Discussed in https://github.com/pybamm-team/PyBaMM/discussions/2521
Google Colab has updated to Python 3.8 but now fails to install PyBaMM due to pybtex.

<div type='discussions-op-text'>

<sup>Originally posted by **reenubosepybamm** December  5, 2022</sup>
%pip install pybamm -q    # install PyBaMM if it is not installed
import pybamm
import numpy as np
import matplotlib.pyplot as plt
GIVING AN ERROR
PluginNotFound                            Traceback (most recent call last)
[<ipython-input-5-103384696aab>](https://localhost:8080/#) in <module>
      2 get_ipython().run_line_magic('pip', 'install pybamm --no-cache -q')
      3 get_ipython().run_line_magic('pip', 'install pybamm -q    # install PyBaMM if it is not installed')
----> 4 import pybamm
      5 import numpy as np
      6 import matplotlib.pyplot as plt

6 frames
[/usr/local/lib/python3.8/dist-packages/pybtex/plugin/__init__.py](https://localhost:8080/#) in _load_entry_point(group, name, use_aliases)
     79         for entry_point in pkg_resources.iter_entry_points(search_group, name):
     80             return entry_point.load()
---> 81     raise PluginNotFound(group, name)
     82 
     83 

PluginNotFound: plugin pybtex.database.input.bibtex not found</div>

Saransh-cpp, 2022-12-05:
This is a known `pybtex` issue, but till now, it was only affecting the `pybamm.print_citations` statements (seeing the error on import is new for me).

Reported last year here - https://github.com/pybamm-team/BattBot/issues/47
Reported in `pybtex` here - https://bitbucket.org/pybtex-devs/pybtex/issues/148/error-pluginnotfound-plugin (I did not follow up. I will post a follow-up comment there.)

The error comes from - 
https://github.com/pybamm-team/PyBaMM/blob/c17756f1b7605403e8e3f025c9d3568517941529/pybamm/citations.py#L149

because the `__init__` method has - 
https://github.com/pybamm-team/PyBaMM/blob/c17756f1b7605403e8e3f025c9d3568517941529/pybamm/citations.py#L38-L39

Maybe we can execute these lines only when a user is not on `Colab` and throw a warning about citations when a user is on `Colab` (till this issue is fixed).

tinosulzer, 2022-12-05:
Thanks @Saransh-cpp , sounds like a good workaround

reenubosepybamm, 2022-12-05:
currently i am working in pybamm i,should i wait for the the bug to be cleared.or should i go to comsol

closed 2022-12-05 by tinosulzer

Saransh-cpp, 2022-12-06:
@reenubosepybamm you should be able to install PyBaMM's latest version in your local environment (not Google Colab) without any issues.

This issue has also been fixed and you can install the develop version of PyBaMM on Google Colab (till the time a new release is made) -
```
pip install git+https://github.com/pybamm-team/PyBaMM@develop
```
instead of -
```
pip install pybamm
```

renatorod, 2022-12-12:
I have been using PyBaMM for most of 2022 with no issues, but since Google Colab was updated to Python 3.8, I have not been able to run any simulations. 

I have tried the following fixes to no avail:
1. Installed PyBaMM on Colab with `pip install git+https://github.com/pybamm-team/PyBaMM@develop` as suggested by @Saransh-cpp 
- Allows installing/importing  pybamm, but throws an error when defining the model or the parameter_values
- `model = pybamm.lithium_ion.DFN(options={"thermal":"lumped"})` or `parameter_values = pybamm.ParameterValues('Chen2020')`
- Error1: KeyError: 'Not a bibtex citation or known citation: Ai2019'
- Error2: KeyError: 'Not a bibtex citation or known citation: Chen2020'
2. Attempted installation on my local environment (desktop) with `pip install pybamm` 
- Throws an ERROR during installation 
- ERROR: PluginNotFound: plugin pybtex.database.input.bibtex not found
3. Installed PyBaMM on my local environment (desktop) with `pip install git+https://github.com/pybamm-team/PyBaMM@develop`
- Allows installation but throws an ERROR during `import pybamm`
- ERROR: PluginNotFound: plugin pybtex.database.input.bibtex not found

Any other suggestions?

PR #2526 (2022-12-05): Fix colab and pybtex issue
# Description

This looks like this right now -

[image]

Now that I think of it, it might be better to not show a warning on importing pybamm (would be too noisy). Rather, we could do -

```py
    def __init__(self):
        # Set of citation keys that have been registered
        self._papers_to_cite = set()

        # Dict mapping citations keys to BibTex entries
        self._all_citations: dict[str, str] = dict()

        if "google.colab" not in sys.modules:
            self.read_citations()
            self._reset()
```
and then -
```py
def print_citations(fil [... cut]