pybamm-team/PyBaMM #2534: [Bug]: Error with EntryPoints

brosaplanella, 2022-12-08:
### PyBaMM Version

22.11

### Python Version

3.9

### Describe the bug

[image]

Two users have brought up to my attention that they get this error when trying to run their scripts in PyBaMM 22.11 (didn't occur in 22.10). The scripts are very standard, and both users use Spyder so I suspect the issue is related to that (I couldn't reproduce the error neither in Windows and Linux, but I am using VS Code).

### Steps to Reproduce

Run a script with PyBaMM 22.11 in Spyder.

tinosulzer, 2022-12-08:
I tried to download Spyder on Mac to test this but it's not working. Can you download Spyder and try?
I'd also be interested to know what the type of `self.__load_entry_point__(key)` is

brosaplanella, 2022-12-09:
```
Traceback (most recent call last):

  File M:\Leranpython\untitled3.py:11
    parameter_values = pybamm.ParameterValues("Chen2020")

  File ~\Anaconda3\lib\site-packages\pybamm\parameters\parameter_values.py:81 in __init__
    if isinstance(values, str) and values in pybamm.parameter_sets:

  File ~\Anaconda3\lib\_collections_abc.py:769 in __contains__
    self[key]

  File ~\Anaconda3\lib\site-packages\pybamm\parameters\parameter_sets.py:51 in __getitem__
    return self.__load_entry_point__(key)()

TypeError: 'EntryPoint' object is not callable
```

Just for completeness, here is the error message as text. I will try to reproduce it in my machine.

brosaplanella, 2022-12-09:
Ok, that's very weird. I used Spyder to try and reproduce the error and it runs fine for me with PyBaMM 22.11.

tinosulzer, 2022-12-14:
Can you test with v22.11.1?

MichaPhilipp, 2022-12-15:
For me still the same error with v22.11.1

francmst, 2022-12-21:
Dears,

I am using v22.11.1 (installed on a machine) with Python 3.9.5. My IDE is Spyder 5. I am running into the same problem. This verifies for all the files where a full set of parameter_values is invoked. NB: files that does not require so (e.g. initial tutorial 10 and 11) run regularly.

For example, I copy/paste the traceback obtained when running Tutorial 1 (initial tutorial):

> runfile('C:/Pybamm_material/Init_tutorials/Pybamm_tutorial_1.py', wdir='C:/Pybamm_material/Init_tutorials')
Traceback (most recent call last):

  File "C:\Pybamm_material\Init_tutorials\Pybamm_tutorial_1.py", line 15, in <module>
    sim = pybamm.Simulation(model)

  File "C:\Users\MF\AppData\Local\Programs\Python\Python39\lib\site-packages\pybamm\simulation.py", line 72, in __init__
    self.parameter_values = parameter_values or model.default_parameter_values

  File "C:\Users\MF\AppData\Local\Programs\Python\Python39\lib\site-packages\pybamm\models\full_battery_models\lithium_ion\base_lithium_ion_model.py", line 83, in default_parameter_values
    return pybamm.ParameterValues("Marquis2019")

  File "C:\Users\MF\AppData\Local\Programs\Python\Python39\lib\site-packages\pybamm\parameters\parameter_values.py", line 81, in __init__
    if isinstance(values, str) and values in pybamm.parameter_sets:

  File "C:\Users\MF\AppData\Local\Programs\Python\Python39\lib\_collections_abc.py", line 768, in __contains__
    self[key]

  File "C:\Users\MastropierroF\AppData\Local\Programs\Python\Python39\lib\site-packages\pybamm\parameters\parameter_sets.py", line 51, in __getitem__
    return self.__load_entry_point__(key)()

TypeError: 'EntryPoint' object is not callable

minatouma, 2022-12-22:
I have the same problem, can't find any solution. Any help or suggestion thanks 

runfile('C:/Users/e179109/untitled0.py', wdir='C:/Users/e179109')
Traceback (most recent call last):

  File "C:\Users\e179109\Anaconda3\lib\site-packages\spyder_kernels\py3compat.py", line 356, in compat_exec
    exec(code, globals, locals)

  File "c:\users\e179109\untitled0.py", line 16, in <module>
    sim = pybamm.Simulation(model, experiment=experiment)

  File "C:\Users\e179109\Anaconda3\lib\site-packages\pybamm\simulation.py", line 72, in __init__
    self.parameter_values = parameter_values or model.default_parameter_values

  File "C:\Users\e179109\Anaconda3\lib\site-packages\pybamm\models\full_battery_models\lithium_ion\base_lithium_ion_model.py", line 83, in default_parameter_values
    return pybamm.ParameterValues("Marquis2019")

  File "C:\Users\e179109\Anaconda3\lib\site-packages\pybamm\parameters\parameter_values.py", line 81, in __init__
    if isinstance(values, str) and values in pybamm.parameter_sets:

  File "C:\Users\e179109\Anaconda3\lib\_collections_abc.py", line 769, in __contains__
    self[key]

  File "C:\Users\e179109\Anaconda3\lib\site-packages\pybamm\parameters\parameter_sets.py", line 51, in __getitem__
    return self.__load_entry_point__(key)()

TypeError: 'EntryPoint' object is not callable

tinosulzer, 2022-12-22:
What does the following script print?
```python3
import importlib_metadata

for entry_point in importlib_metadata.entry_points(group="pybamm_parameter_sets"):
    print(type(entry_point))
    print(isinstance(entry_point, importlib_metadata.EntryPoint))
```

francmst, 2022-12-23:
Output from Spyder (pybamm installed on machine) and output from jupyter (on virtual environment after typing "import pybamm")

<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False
<class 'setuptools._vendor.importlib_metadata.EntryPoint'>
False

Output from jupyter (on virtual environment typing the script without any prior command):
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True
<class 'importlib_metadata.EntryPoint'>
True

tinosulzer, 2022-12-28:
Thanks, this will help to fix the problem, could you run this to make sure the fix will work
```python3
import importlib_metadata
import setuptools._vendor.importlib_metadata as setuptools_metadata

for entry_point in importlib_metadata.entry_points(group="pybamm_parameter_sets"):
    print(
        isinstance(
            entry_point, (importlib_metadata.EntryPoint, setuptools_metadata.EntryPoint)
        )
    )
```

PR #2584 (2022-12-28): #2534 fix spyder entry points
# Description

Fixes #2534 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ flake8`
- [ ] All tests pass: `$ python run [... cut]
closed 2022-12-28 by tinosulzer