pybamm-team/PyBaMM #2538: Reference Performance Test (RPT) experiment

brosaplanella, 2022-12-09:
### Description

The capacity of the battery (e.g. when doing ageing tests) is typically done through RPTs, an experimental step consisting of a full charge and a CC discharge at a given C-rate until the cut-off voltage is hit. The capacity in that discharge is the "measured capacity".

These tests can be easily run with the current features (though the experiment needs to be split and solution reinitialised which can be cumbersome). However, this is a fairly standard test which is intercalated between many other steps, and often we only care about the measurements during the RPT. Possible ways around that would be to define a separate RPT class and then:
* Store RPT subsolutions (or maybe even just the capacity measure) as a separate variable in the solution.
* Alternatively, just store a list with the indices corresponding the RPT so they can be easily accessed.

Anyway, what are people thoughts on that?

### Motivation

Fairly standard experiment type used in ageing experiments.

### Possible Implementation

Define a separate RPT experimental step (ties with #2322)

### Additional context

This might make #2219 no longer needed, as the main motivation was doing RPT.

tinosulzer, 2022-12-09:
I'm a little bit reluctant to add more special cases to the Simulation class which is already a bit of a god class. Isn't this fairly easy to do using `solution.first_state` like in your SPMe-SR repo?

brosaplanella, 2022-12-12:
Thinking more carefully on this, an option would be to "tag" specific steps (e.g. adding `[RPT]` at the end of the string). Then we can just have a `search` method that returns the cycles that match the tag (this would allow a lot more flexibility).

tinosulzer, 2022-12-12:
agreed - tagging (/labeling) is part of what was suggested by @Scottmar93 in #2322

brosaplanella, 2023-01-05:
I will start working on tagging experiments. @Scottmar93 let me know if that conflicts with your workflow for #2322.

closed 2023-01-06 by brosaplanella

Scottmar93, 2023-01-08:
@brosaplanella Should be fine as I think the idea is to make a separate class for more customizable cases in order to avoid breaking existing experiment class functionality.

PR #2593 (2023-01-06): Add tags to `Experiment`
# Description

Tags can be added to the end of an experiment as `[tag1,tag2]`. I have added the method `search_tag` which returns the list of cycles with a certain tag, which should help saving only certain cycles.

Fixes #2538 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Opti [... cut]