pybamm-team/PyBaMM #2553: Cell-level State of Charge

tinosulzer, 2022-12-14:
We do not currently provide "State of Charge" as a variable, because it's not clear how to calculate this at the cell level:
- Discharged capacity divided by nominal cell capacity:
  - easy to calculate from existing parameters
  - what is the initial soc?
  - what if nominal cell capacity is wrong (i.e. not equal to the amount of charge that can be passed between voltage limits)?
- Calculation based on min/max electrode stoichiometry and nominal cell capacity
  - requires specifying "Stoichiometry at 100% SOC" in each electrode instead of "Initial concentration"
  - SOC can be calculated as e.g. `(x - x_100)`
     - what if nominal cell capacity is wrong? 
- Calculation based on electrode SOH
   - requires a pre-calculation step to calculate electrode SOH before solving the model
   - after that

## Proposed solution

1. Add a "Calculate SOC" model option which can be "true" or "false"
2. If true, add simple SOC submodel based , and perform `eSOH` calculation in `ParameterValues` class to define the parameters for the SOC submodel

Caveat: this will be wrong as the cell degrades. We can figure out updating parameter values later

tinosulzer, 2022-12-14:
This will finally enable the much-requested feature "Charge until 98% SOC"

brosaplanella, 2022-12-14:
Nice! To add to this, we typically calculate SoC as discharged capacity / RPT calculated capacity (or one minus that, to be precise). This way you can update it as the battery ages, but it gets trickier to calculate as it requires a separate simulation.

tinosulzer, 2022-12-14:
I think all of the following are equivalent:
1. `soc = initial_soc - Q_Ah / Q`
2. `soc = (x-x0)/(x100-x0)`
3. `soc = (y-y0)/(y100-y0)` (both numerator and denominator are negative)

So any way we do it we have to pre-calculate something (`initial_soc`/`Q`/`x0`/`x100`/`y0`/`y100`). The bad news is we can't trust the value the user enters since it might be incompatible with the voltage limits and hence lead to SOC<0 or SOC>1. The good news is that the electrode SOH model can do the calculation for us in a way that guarantees that the SOC is always in [0,1] given the parameter values and voltage limits.

> it gets trickier to calculate as it requires a separate simulation

The capacity calculated by the "electrode SOH" model should give the same answer as the RPT without requiring a whole simulation (just solving an algebraic model) but let me know if you have a case where it doesn't

brosaplanella, 2022-12-14:
Doesn't "electrode SOH" give the theoretical capacity (i.e. infinitely slow discharge)? RPT is usually at a given non-negligible C-rate (we typically use C/3). It should be possible write an analytical approximation for it using the SPMe model.

tinosulzer, 2022-12-14:
Yes but in terms of calculating SOC we want the theoretical capacity, right? Since the SOC at the end of the C/3 discharge RPT should not be zero. I agree that the C/3 RPT is required to record the "C/3 capacity" at various cycles, but not to calculate the SOC

brosaplanella, 2022-12-14:
Not so sure... We do it based on RPT (not sure about other people) and I think the nominal capacity is defined on a similar way. might be good to ask around? In any case, we can just go with theoretical capacity for the time being and we can always extend SOH models to account for RPT.

tinosulzer, 2022-12-14:
Yes would be good to get more input on this. We can also have multiple "SOC" variables based on different capacity calculations

Scottmar93, 2023-01-08:
> Not so sure... We do it based on RPT (not sure about other people) and I think the nominal capacity is defined on a similar way. might be good to ask around? In any case, we can just go with theoretical capacity for the time being and we can always extend SOH models to account for RPT.

I think using the RPT at 25oC to define the SoC is what's usually used with doing battery testing. The OCV at 100% SoC is taken to be the rested voltage after the CCCV charge and the OCV at 0% SoC is taken to be rested voltage after the CC discharge. 

I guess what's annoying about this definition of SoC is that it's dependent upon the cell resistance so if the cell ages then there will be different 100% and 0% OCV values.... Also it can only be calculated by actually running the RPT...

jaskiratsingh2000, 2023-02-20:
Hi! Is its still open to work? I can explore though.

iamshivamyadav, 2023-09-28:
> Yes would be good to get more input on this. We can also have multiple "SOC" variables based on different capacity calculations

As per my understanding following steps can be taken 

1. Charge the cell upto upper cut off voltage and relaxation rest period more than 3 hours which will give OCV 100 % SoC.

2. Perform Pulse Dicharge Test for every 10 % SOC discharge and relaxation period more than 3 hours which will give OCVs at different SOC like 90 %, 80% ,70 etc.

3. Perform above step 1 and step 2 at different temperatures which will give SOC and OCV at different temperatures 

4. If possible perform step 1, 2 and 3 for new, 500 cycle cell, 1000 cycle cell and more aged cell. It will help to calculate SOH by estimating internal resistance from the Pulse Discharge Test voltage curve graph.

rifat37dlr, 2024-02-29:
> I think all of the following are equivalent:
>
> 1. `soc = initial_soc - Q_Ah / Q`
> 2. `soc = (x-x0)/(x100-x0)`
> 3. `soc = (y-y0)/(y100-y0)` (both numerator and denominator are negative)
>
> So any way we [...]

Could anyone please tell me what's the meaning of x and y?

tinosulzer, 2024-03-06:
x is stoichiometry in the negative electrode (goes from 0 to 1), y in the positive electrode