pybamm-team/PyBaMM #2556: Internal Input Parameter

tinosulzer, 2022-12-14:
Proposal: add an "internal_only" boolean flag to input parameters. Internal input parameters should read their value from a dictionary `model._input_parameters` that should not be accessed by users. This would be useful to handle parameters that need to be updated internally without rebuilding the model, but aren't "input parameters" in the sense that they aren't (can't be) passed in by users. Some examples of these are those related to experiments (e.g. `start_time`), and this will also be useful for the SOC limits in #2553 

@abillscmu this is an easy way to do what you call `PseudoInputParameter`, what are your thoughts? The difference is that this approach only lets you make the internal input parameters a fixed scalar, whereas the `PseudoInputParameter` could change it to any expression tree

abillscmu, 2022-12-14:
From the standpoint of should there exist something like this, yes I think there should.

However, I find the implementation of `PseudoInputParameter` very convenient:

(1) As it's own class, it's nice to be able to define methods like `set_pseudo` which is used to set its "value" and which don't make sense for a standard input parameter.

(2) The fact that it can be an expression tree is a key enabler of certain features of the .jl code. For example, if there are nested models, it's nice for the "upstream" model to be able to determine the inputs for the "downstream model". 

(3) The combination of (1) and (2) also gives flexibility: by having a parametric method for handling the code at "compile-time", there is flexibility in when and how the upstream model computes the parameter. For example, if the downstream model is an external library, the upstream model can explicitly compute the parameter and pass it into a call to the external library. Actually the opposite is true as well: the upstream model could be an external library and pass in the parameter to the downstream (pybamm) model. On the other hand, if both models are pybamm models, the `PseudoInputParameter` can act as a passthrough and never appear explicitly in the final function.

I'm not sure if this is relevant for anything other than the specific use case for which I wrote `PseudoInputParameter`, but those are my thoughts.

tinosulzer, 2022-12-14:
Ok got it. The disadvantage of `PseudoInputParameter` for PyBaMM is that it requires re-building the casadi model - so if there are internal inputs we want to change every cycle it's inefficient. Sounds like we should have both

abillscmu, 2022-12-14:
I think having both makes sense.

aabills, 2024-11-22:
I don't think so, I think `CoupledVariable` should take care of this.