pybamm-team/PyBaMM #2562: Separate Voltages for events and setting initial SOC

js1tr3, 2022-12-15:
### Description

It appears the parameters 
Lower voltage cut-off [V]	2.8
Upper voltage cut-off [V]	4.2

are used both to set the initial SOC and for simulation events, as a result if the voltage limits for discharge are changed, this impacts the initial SOC. 

### Motivation

It would be good to have a sperate variable for the simulation event termination and calculation of SOC/initial stoichiometry.
The present implementation is breaking my use case of comparing the model to different discharge voltages, as the initial SOC changes also.

tinosulzer, 2022-12-15:
Hi Jason :) You can change the values in the experiment specification
```python3
pybamm.Experiment(
    [
        "Discharge at C/5 for 10 hours or until 3 V",
        "Rest for 1 hour",
        "Charge at 1 A until 4.2 V",
        "Hold at 4.2 V until 10 mA",
        "Rest for 1 hour",
    ]
)
```
without affecting the voltage limits for SOC, is this what you need?

js1tr3, 2022-12-15:
Hi Tino, thanks for the rapid response! I'm actually using a recorded
current trace, and feeding in using the old interpolant method. I'll give
the new experiment using drive cycles a go. I just pulled upstream changes
and broke a few things on my branch... so im trying to catch up on
parameter changes.

What I want to do is charge to 110% SOC or over discharge assuming the ocp
curve is well defined and stoichiometries are nonzero (within limits) this
means the voltage events should be outside the min/max voltages used for
calculating SOC... if that makes sense... happy to talk through my idea if
you have time.

Best -J

tinosulzer, 2022-12-15:
I see, that makes sense. And even if the SOC is still <100% the voltage might go above the cut-off. So we should add parameters "Open circuit potential at 0% SOC" and "Open circuit potential at 100% SOC", which can be different from the voltage cut-offs.

Could you open a PR to do this? It should be pretty straightforward. Basic steps are:
1. Add these parameters to each parameter set, defaulting to the same value as the cut-offs
2. Add a two new `Parameter` object to the `lithium_ion_parameters.py` files (e.g. `self.opc_soc_0`, `self.ocp_soc_100`)
3. Replace the cut-off voltage in this line (and other places in that file) with the new target ocp voltage https://github.com/pybamm-team/PyBaMM/blob/1f25a1a3bbaf739848b67dec3b81562db5ad5c7a/pybamm/models/full_battery_models/lithium_ion/electrode_soh.py#L446

js1tr3, 2022-12-15:
I'll take a stab at it!

js1tr3, 2022-12-17:
I made the changes you suggested in my repo for testing, but it seems that
the voltage is not updated,
is it possible that if the initial SOC is still one then the calculation of
the initial concentrations is being skipped?

https://github.com/js1tr3/PyBaMM/blob/7b943d147f14a51892e29cf1b6e16295ec1358b7/pybamm/simulation.py#L384

tinosulzer, 2022-12-19:
Does `test_electrode_soh.py` work as expected?

tinosulzer, 2022-12-20:
Can you open a draft pull request so I can check it? It shouldn't make a difference what Vmin and Vmax are for this calculation, if it's based entirely on OCP at 0/100% SOC. Make sure you change `voltage_low_cut_dimensional` to `opc_soc_0_dimensional` everywhere relevant.

rtimms, 2023-05-15:
@js1tr3 how is this going?

js1tr3, 2023-05-15:
> @js1tr3 how is this going?

Good, got tied up with teaching this semester, but hope to get back on this shortly. Any suggestions for resolving conflicts in jupyter notebooks?

tinosulzer, 2023-10-04:
Solved by #2580

closed 2023-10-04 by tinosulzer