pybamm-team/PyBaMM #2573: [Bug]: Parameter Set "Mohtat2020" have Initial Condition Problem

1836005678, 2022-12-20:
### PyBaMM Version

22.11.1+22.11.1

### Python Version

3.8(Online Colab)+3.9(Local VSCode)

### Describe the bug

According to Pybamm Document (https://pybamm.readthedocs.io/en/latest/source/parameters/parameter_sets.html), there are 14  parameter sets for use (['Ai2020', 'Chen2020', 'Chen2020_composite', 'ECM_Example', 'Ecker2015', 'Marquis2019', 'Mohtat2020', 'NCA_Kim2011', 'OKane2022', 'ORegan2022', 'Prada2013', 'Ramadass2004', 'Sulzer2019', 'Xu2019']), while 12 of them are for DFN model (['Ai2020', 'Chen2020', 'Chen2020_composite', 'Ecker2015', 'Marquis2019', 'Mohtat2020', 'NCA_Kim2011', 'OKane2022', 'ORegan2022', 'Prada2013', 'Ramadass2004', 'Xu2019']), in which only 9 of them can work well in my program.
Unfortunately, the set 'Mohtat2020' can not work well, as to it comes to the model event "['Minimum voltage'] are non-positive at initial conditions". I have changed the parameter 'Current function [A]' to other values rather than the default 5A, but the event still happened.
Please tell me how to deal with this problem. Thank you very much!

### Steps to Reproduce

Python Code:
import pybamm
parameter_values = pybamm.ParameterValues("Mohtat2020")
model = pybamm.lithium_ion.DFN()
sim = pybamm.Simulation(model, parameter_values=parameter_values)
sim.solve([0, 3600])
sim.plot()

### Relevant log output

```shell
Online Colab log output:
---------------------------------------------------------------------------
SolverError                               Traceback (most recent call last)
<ipython-input-14-c8081af8ccdc> in <module>
      1 model = pybamm.lithium_ion.DFN()
      2 sim = pybamm.Simulation(model, parameter_values=parameter_values)
----> 3 sim.solve([0, 3600])
      4 sim.plot()

2 frames
/usr/local/lib/python3.8/dist-packages/pybamm/simulation.py in solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, callbacks, **kwargs)
    615                         )
    616 
--> 617             self._solution = solver.solve(self.built_model, t_eval, **kwargs)
    618 
    619         elif self.operating_mode == "with experiment":

/usr/local/lib/python3.8/dist-packages/pybamm/solvers/base_solver.py in solve(self, model, t_eval, external_variables, inputs, initial_conditions, nproc, calculate_sensitivities)
    841 
    842         # Check initial conditions don't violate events
--> 843         self._check_events_with_initial_conditions(
    844             t_eval_dimensionless, model, ext_and_inputs_list[0]
    845         )

/usr/local/lib/python3.8/dist-packages/pybamm/solvers/base_solver.py in _check_events_with_initial_conditions(self, t_eval, model, inputs_dict)
   1058             idxs = np.where(events_eval < 0)[0]
   1059             event_names = [termination_events[idx].name for idx in idxs]
-> 1060             raise pybamm.SolverError(
   1061                 f"Events {event_names} are non-positive at initial conditions"
   1062             )

SolverError: Events ['Minimum voltage'] are non-positive at initial conditions

Local VSCode log output:
Traceback (most recent call last):
  File "d:\WPS\3-毕业设计\4-代码\1-敏感度分析\param_set_debug.py", line 6, in <module>
    sim.solve([0, 3600])
  File "C:\Users\MGY\AppData\Local\Programs\Python\Python39\lib\site-packages\pybamm\simulation.py", line 617, in solve
    self._solution = solver.solve(self.built_model, t_eval, **kwargs)
  File "C:\Users\MGY\AppData\Local\Programs\Python\Python39\lib\site-packages\pybamm\solvers\base_solver.py", line 843, in solve
    self._check_events_with_initial_conditions(
  File "C:\Users\MGY\AppData\Local\Programs\Python\Python39\lib\site-packages\pybamm\solvers\base_solver.py", line 1060, in _check_events_with_initial_conditions
    raise pybamm.SolverError(
pybamm.expression_tree.exceptions.SolverError: Events ['Minimum voltage'] are non-positive at initial conditions
```

1836005678, 2022-12-20:
I find that I have used the wrong model.
This parameter set is for MPMe model according to the reference.
However, the MPM model still can not work well using this parameter set, as the set miss the item "Negative area-weighted particle-size distribution [m-1]".

Log files:
KeyError: "'Negative area-weighted particle-size distribution [m-1]' not found. Best matches are ['Negative particle radius [m]', 'Negative electrode diffusivity [m2.s-1]', 'Negative current collector conductivity [S.m-1]']"

Item 'Negative particle radius [m]' equals to 2.5e-06.
However, in the original reference, it is indeed provided in the form of distribution function, thus there are 10% / 22% / 30% / 25% / 13% particles , whose radius equal to 1.5e-06 / 2.0e-06 / 2.5e-06 / 3.0e-06 / 3.7e-06 m.
The parameter set may need to be adjusted.

tinosulzer, 2022-12-20:
First post: For the Mohtat2020 parameter set you have to start with a charge since the parameters are given for 0% SOC. We are working on standardizing the parameter sets to all start at 100% SOC.

Second post: indeed, the Mohtat2020 parameter set comes from an MPMe paper, but it still works with DFN. To add the relevant parameters for MPM to any parameter set, do
```python3
parameter_values = pybamm.get_size_distribution_parameters(parameter_values)
```

1836005678, 2023-05-04:
First post:  
100% SOC Parameter value solved by `sim.build(initial_soc=1)` for Mohtat2020
`"Initial concentration in negative electrode [mol.m-3]": 23956.1769437291,`
`"Initial concentration in positive electrode [mol.m-3]": 1186.8414661132,`
Maybe helpful for developers.