pybamm-team/PyBaMM #2607: Interface docs need updating

rtimms, 2023-01-17:
The [docs for the interface submodels](https://pybamm.readthedocs.io/en/latest/source/models/submodels/interface/index.html) should match the folder structrure [here](https://github.com/pybamm-team/PyBaMM/tree/develop/pybamm/models/submodels/interface).

dikwickley, 2023-01-24:
hello @rtimms I would like to work on this issue, can you please assign it to me?

rtimms, 2023-01-24:
Thanks @dikwickley ! Let me know if you have any questions.

Yash-Sajwan24, 2023-01-24:
Hello, @rtimms 
I would like to work on this issue, can you please assign it to me? I am a beginner and looking forward to contributing.

Yash-Sajwan24, 2023-01-24:
Can you please specify the order as most of the names of folders do not match with the doc?

rtimms, 2023-01-25:
Thanks @Yash-Sajwan24!

The docs should follow the folder structure, e.g. 

- BaseInterface
- TotalInterfacialCurrent
- Open Circuit Potential 
    - BaseOpenCircuitPotential
    - CurrentSigmoidOpenCircuitPotential
    - SingleOpenCircuitPotential
- Kinetics
    - ...
- ...   

@dikwickley @Yash-Sajwan24 can you coordinate between yourselves who is going to work on this to avoid duplication of effort?

Yash-Sajwan24, 2023-01-25:
I still don't get it .☹️ 
[image]
[image]

rtimms, 2023-01-25:
The [interface docs](https://github.com/pybamm-team/PyBaMM/tree/develop/docs/source/models/submodels/interface) need to be modified to match the file structure in the [interface submodel directory](https://github.com/pybamm-team/PyBaMM/tree/develop/pybamm/models/submodels/interface). You will need to add extra folders and .rst files.

Take a look at how the [electrode docs](https://github.com/pybamm-team/PyBaMM/tree/develop/docs/source/models/submodels/electrode) are set up based on the [electrode submodels directory](https://github.com/pybamm-team/PyBaMM/tree/develop/docs/source/models/submodels/electrode).

dikwickley, 2023-01-26:
@rtimms i have made a new interface folder and have completed kinetics and Interface Utilisation.
Here is the demo: https://dikwickley-pybamm.readthedocs.io/en/latest/source/models/submodels/interface2/
let me know if it's okay

rtimms, 2023-01-26:
Looks good, thanks! Please open a PR

dikwickley, 2023-01-26:
@rtimms please review the PR. if there are any other modules that require documentation updates let me know. thanks :)

PR #2624 (2023-05-03): Interface docs update
# Description

Updated the docs for the interface module.

Fixes #2607 

## Type of change
documentation update

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ flake8`
- [ ] All tests pass: `$ python run-tests.py --unit`
- [ ] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tests.py --quick`.

## Further checks:

- [ ] [... cut]
closed 2023-05-03 by rtimms