pybamm-team/PyBaMM #2611: NumFOCUS onboarding

tinosulzer, 2023-01-18:
- [ ] Add the “Powered by NumFOCUS” badge for your website and GitHub.
Markdown for adding NumFOCUS badge to project websites: [![Powered by
NumFOCUS]( https://img.shields.io/badge/powered%20by-NumFOCUS-orange.s
vg?style=flat&colorA=E1523D&colorB=007D8A )](http://numfocus.org)
- [ ] Add the Fiscal Sponsor Readme Attribution to your project’s GitHub:
https://github.com/numfocus/templates/blob/master/fiscal-sponsor-readme-attribution.md
- [ ] Add the NumFOCUS sponsored project logo ( with link to numfocus.org ) to your
project’s website (under About) and language indicating that you are a sponsored project
of NumFOCUS. For example: “[Project] is a fiscally sponsored project of NumFOCUS, a
nonprofit dedicated to supporting the open-source scientific computing community. If you
like [Project] and want to support our mission, please consider making a [donation](link
to Salsa campaign) to support our efforts.” (*We will provide you the campaign link to
include on your website once you’ve sent us the marketing materials listed above)
- [ ] Link to the donation page. Please use this exact language alongside the donation link:
“NumFOCUS is a 501(c)(3) non-profit charity in the United States; as such, donations to
NumFOCUS are tax-deductible as allowed by law. As with any donation, you should
consult with your personal tax adviser or the IRS about your particular tax situation.”

tinosulzer, 2023-01-23:
For the last point, I think the sponsorship page is https://numfocus.org/donate-for-pybamm

dikwickley, 2023-01-24:
hello @tinosulzer I would like to work on this issue. Can you please assign it to me?

brosaplanella, 2023-01-24:
Hi @dikwickley! We already have an ongoing PR for this #2614. If you want to contribute to PyBaMM a great place to start is by tackling one of the [issues labelled as easy](https://github.com/pybamm-team/PyBaMM/issues?q=is%3Aopen+is%3Aissue+label%3A%22difficulty%3A+easy%22).

closed 2023-02-17 by tinosulzer