pybamm-team/PyBaMM #2619: [Bug]: Intermittent pulse plot seems unphysical

nbiju1998, 2023-01-26:
### PyBaMM Version

22.12

### Python Version

3.9.10

### Describe the bug

Hello,

I am wanting to simulate an intermittent pulse experiment on the SPMe model. I am combining this with the lumped thermal model and using parameters from 'Marquis 2019.' I apply a load of 0.5 C for 5 minutes and let the cell rest for 1 hour. I repeat this until the cutoff voltage is reached. Instead of the voltage decreasing, it reaches a plateau. The cell is continually being pulsed intermittently but the discharge capacity and the voltage remain constant. I have attached a figure of the two down below:

[image]
[image]

### Steps to Reproduce

import pybamm

options = {"thermal": "lumped"}
param = pybamm.ParameterValues("Marquis2019")
param.update({"Total heat transfer coefficient [W.m-2.K-1]": 1.0})

experiment = pybamm.Experiment(
    [
        ("Discharge at 0.5C for 0.08 hours or until 3.63 V (1 second period)",
        "Rest for 1 hour (1 second period)",
        )
    ]
    * 40,
)
model = pybamm.lithium_ion.SPMe(options)
sim = pybamm.Simulation(model, experiment=experiment, solver=pybamm.CasadiSolver())

brosaplanella, 2023-01-26:
Hi! I think the behaviour is physical. As you get closer to the cut-off voltage (3.63 V in this case) the pulse is shorter (it doesn't last 0.08 hours, it stops when the cut-off voltage is hit). Therefore, the capacity plateaus as each pulse extracts less and less capacity. Similarly, the voltage at rest plateaus.

nbiju1998, 2023-01-27:
Oh okay, I see, thank you!

closed 2023-01-27 by nbiju1998