pybamm-team/PyBaMM #2622: How to implement a constant resistance experiment

skywo1f, 2023-01-26:
I have this setup:
```
import pybamm
param = pybamm.ParameterValues("Mohtat2020")
exp = pybamm.Experiment(
    [       
            "Discharge at 1 C for 0.5 hours", 
            "Rest for 5 minutes",
      
    ]
)
model = pybamm.lithium_ion.SPM(
        {
            "operating mode": "resistance"
            }
)
sim = pybamm.Simulation(model,experiment=exp,parameter_values=param)
sim.solve([0, 3600], initial_soc=1)
solution = sim.solution
sim.plot(["Current [A]", "Terminal voltage [V]", "Discharge capacity [A.h]"])
```
but it gives the following error:
```
Traceback (most recent call last):
  File "/home/iviti/anaconda3/lib/python3.9/site-packages/pybamm/parameters/parameter_values.py", line 582, in process_symbol
    return self._processed_symbols[symbol]
KeyError: FunctionParameter(-0x3a348ce205434d50, Resistance function [Ohm], children=['time * 96485.33212 * Maximum concentration in negative electrode [mol.m-3] * (Negative electrode thickness [m] + Separator thickness [m] + Positive electrode thickness [m]) / absolute(Typical current [A] / (Number of electrodes connected in parallel to make a cell * Electrode width [m] * Electrode height [m]))'], domains={})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/iviti/anaconda3/lib/python3.9/site-packages/pybamm/util.py", line 58, in __getitem__
    return super().__getitem__(key)
KeyError: 'Resistance function [Ohm]'

```
Any idea why the error is occurring, and how to change the experiment portion to simulate a constant resistance run instead of a constant current run?

tinosulzer, 2023-01-26:
You need to add the resistance function as a parameter, e.g.
```python3
parameter_values.update({"Resistance function [Ohm]": 4})
```

skywo1f, 2023-01-26:
I tried both 
`param.update({"Resistance function [Ohm]": 4})`
(which gives basically the same error) 
and 
`parameter_values.update({"Resistance function [Ohm]": 4})`
(which gives parameter_values is not defined)

Is there a specific place where I have to add that?

tinosulzer, 2023-01-26:
Sorry, it should be
```python3
param.update({"Resistance function [Ohm]": 4}, check_already_exists=False)
```
below where you define `param`

skywo1f, 2023-01-26:
that seemed to run. So then does it just ignore the part where I ask for 3C?

tinosulzer, 2023-01-26:
Yes - resistance control within the experiment is not yet implemented. I'd say it's easy-medium difficulty to do, if you want to try it

closed 2023-01-26 by skywo1f