pybamm-team/PyBaMM #2631: Hackathon 29th March 2023 in Oxford

tinosulzer, 2023-01-27:
We will be hosting an in-person hackathon on 29th March in Oxford.

I﻿n this hackathon, you will have the opportunity to learn how to contribute to PyBaMM, while working on real examples under the expert guidance of members the PyBaMM team. The hackathon will also include an introduction on how to contribute to the project (more details of the agenda will be published closer to the date). Each participant should bring their own laptop and have PyBaMM installed from source (see details [here](https://pybamm.readthedocs.io/en/latest/install/install-from-source.html)). L﻿unch and tea&coffee will be provided.

This workshop is supported by The Faraday Institution `"Multi-Scale Modelling" project [EP/S003053/1 grant number FIRG025].

Important n﻿ote: this hackathon is not an introductory course on how to use PyBaMM. E﻿xperience with Python and PyBaMM is required, and basic experience with git is strongly recommended.

tinosulzer, 2023-01-27:
To do:
- [ ] add this to readme
- [ ] add banner to website

tinosulzer, 2023-01-31:
Thanks! Fixed in both places

brosaplanella, 2023-03-30:
Hackathon took place and went very well, we can now close this issue.

closed 2023-03-30 by brosaplanella