pybamm-team/PyBaMM #2639: Remake pybamm.org with scientific python hugo theme

tinosulzer, 2023-01-30:
Can be hosted as a separate repository on GitHub. We can then host the docs on `docs.pybamm.org`

Theme: https://theme.scientific-python.org/

Examples: https://numpy.org/, https://scipy.org/

brosaplanella, 2023-01-30:
I really like the numpy one with the case studies

jaskiratsingh2000, 2023-02-20:
Thanks @tinosulzer and @brosaplanella for opening this up. I'm currently on it as draftting the proposal on the same.

brosaplanella, 2023-03-04:
Any updates with this? If not, I will create a pybamm.org repo in our team so I can start moving the governance docs (and others) there.

agriyakhetarpal, 2023-05-29:
Just adding a comment here to let it be known that work on this has been started and the source code is in the [pybamm-team/pybamm.org repository](https://github.com/pybamm-team/pybamm.org); please feel free to raise issues, add pull requests, and suggest improvements for the infrastructure there.

brosaplanella, 2023-06-12:
Closing this as the repository for the website has been now set up and discussion can follow there: https://github.com/pybamm-team/pybamm.org

closed 2023-06-12 by brosaplanella