pybamm-team/PyBaMM #2660: Fix broken URLs

Saransh-cpp, 2023-02-04:
URL checker was down temporarily but #2659 brings it back up again.

At the moment, there are a lot of broken URLs - https://github.com/pybamm-team/PyBaMM/actions/runs/4090861650/jobs/7054568267 - which should be fixed.

Most of them originate from the documentation revamp, some of them are false alarms, and some of them are from external repositories/sources.

jeromtom, 2023-02-04:
Started working on it #2656

jeromtom, 2023-02-04:
@Saransh-cpp 
How do we know where these links were used?
Then we can replace it with the correct link like in #2656

Saransh-cpp, 2023-02-04:
You can use the search feature of your IDE (VSCode has a great search functionality).

Alternatively, you can use CLI tools like `grep` (UNIX) and `findstr` (Windows). For example -
```
cd PyBaMM
grep -nr the/url/you/want/to/search     # -r -> recursive, -n -> print line numbers
```

Something similar should work for `findstr` as well.

jeromtom, 2023-02-04:
Thank you for that. Used `findstr` and found the file location of missing links which can be viewed from [Google Sheet-PyBaMM Link Fixes](https://docs.google.com/spreadsheets/d/16ZapQc9CE0XzdqT_ck_mtRR1jOxSDoE27oRlWMNEKfM/edit?usp=sharing)

PR #2656 (2023-02-19): Updated broken links.
# Description
Changes the depreciated/broken link in CONTRIBUTING.md to the correct working link:
https://pybamm.readthedocs.io/en/latest/source/user_guide/installation/install-from-source.html

Fixes #2655 and partially fixes #2660

## Type of change
Documentation Update
- [X] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [X] No style issues: `$ pre-commit run`
- [X] All tests pass: `$ python run-tests.py --unit`
- [X] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tes [... cut]
closed 2023-02-19 by tinosulzer