pybamm-team/PyBaMM #2670: Replace `interp2d` in `ProcessedVariable`

rtimms, 2023-02-08:
We currently use `interp2d` when creating processed variables, but this is deprecated. We should switch to `RegularGridInterpolator`.

```
DeprecationWarning: `interp2d` is deprecated!
`interp2d` is deprecated in SciPy 1.10 and will be removed in SciPy 1.12.0.

For legacy code, nearly bug-for-bug compatible replacements are
`RectBivariateSpline` on regular grids, and `bisplrep`/`bisplev` for
scattered 2D data.

In new code, for regular grids use `RegularGridInterpolator` instead.
For scattered data, prefer `LinearNDInterpolator` or
`CloughTocher2DInterpolator`.
```

jeromtom, 2023-02-12:
2. Alternative to interp2d: regular grid in the [transition guide](https://gist.github.com/ev-br/8544371b40f414b7eaf3fe6217209bff) is relevant to this.

jaskiratsingh2000, 2023-02-19:
HI @rtimms, and @jeromtom Can you let me know that where can I find the processed variable implementation fo "interp2d"? Thanks! :)

rtimms, 2023-02-19:
[Here](https://github.com/pybamm-team/PyBaMM/blob/12cbcff7fe0de9694b4be8e16e47038b729ecde9/pybamm/solvers/processed_variable.py#L630). There may be other places where it appears, so worth searching (e.g. doing ctrl+shift+f in vs code).

jeromtom, 2023-02-19:
It can be found here `pybamm\solvers\processed_variable.py`.

jeromtom, 2023-02-19:
Using [findstr](https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/findstr)  and running `findstr /s /i interp2d *.*`  gives this
[image]

8 mentions of `interp2d`, the first 4 in the documentation, and the last 4 in the code.

tinosulzer, 2023-02-19:
Sites in `sites-packages` are not part of documentation, they are installed dependencies.

I don't know what's in your `build` folder but it looks like it's a duplicate of the repository.

So ignore the first 6, it's only the last two

jaskiratsingh2000, 2023-02-20:
@rtimms , @jeromtom , and @tinosulzer Thanks for letting me know, I have made the changes as a part of pull request abiove. Could you please review it once? Thanks in Advance!

arjxn-py, 2023-04-02:
@rtimms If noone is working on this issue, I'd like to give it a try.

rtimms, 2023-04-03:
Hi @arjxn-py it looks like @jaskiratsingh2000 is no longer working on this, so feel free to pick it up. There is an open PR #2709 that may help. Thanks!

closed 2023-04-30 by tinosulzer

arjxn-py, 2023-04-30:
Thanks @tinosulzer, I was really stuck at this.

tinosulzer, 2023-04-30:
Yeah it was getting a bit messy with all the different interpolants, switching to xarray made it much easier

PR #2865 (2023-04-30): Issue #2670 replace interp2d in processed variable
# Description

replace interp2d in processed variable

Fixes #2670 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No [... cut]