pybamm-team/PyBaMM #2682: [Bug]: Two-phase electrode cycling

BradyPlanden, 2023-02-13:
### PyBaMM Version

23.1

### Python Version

3.8.10

### Describe the bug

Utilising the two-phase electrode definition for cycling with the `pybamm.Experiment` function returns the following error, which seems to be from the "Primary:" and "Secondary:" naming conventional utilised in the two-phase parameter file `Chen2020_composite.py`. 

I think the workaround at the moment is to construct `"Current function [A]"` in such a way as to complete multi-cycle simulations, either through a .CSV interpolant or functionally. Does that seem correct?

### Steps to Reproduce

```
import pybamm

pybamm.set_logging_level("INFO")
# load models
options = {
    "particle phases": ("2", "1"),
    "open circuit potential": (("single", "current sigmoid"), "single"),
}
models = [
    pybamm.lithium_ion.SPM(options),
    pybamm.lithium_ion.SPMe(options),
    pybamm.lithium_ion.DFN(options),
]

parameter_values = pybamm.ParameterValues("Chen2020_composite")

experiment = pybamm.Experiment(
    [
        (
            "Charge at 1 C until 4.2 V",
            "Hold at 4.2 V until C/10",
            "Rest for 5 minutes",
            "Discharge at 1 C until 2.8 V",
            "Rest for 5 minutes",
        )
    ]
    * 2)

# create and run simulations
sims = []
for model in models:
    sim = pybamm.Simulation(model, experiment=experiment, parameter_values=parameter_values)
    sim.solve([0, 9000])
    sims.append(sim)
```

### Relevant log output

```shell
/usr/local/lib/python3.8/dist-packages/pybamm/simulation.py in solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, callbacks, **kwargs)
    623 
    624             # Set up eSOH solver (for summary variables)
--> 625             esoh_solver = self.get_esoh_solver(calc_esoh)
    626 
    627             if starting_solution is None:

/usr/local/lib/python3.8/dist-packages/pybamm/simulation.py in get_esoh_solver(self, calc_esoh)
    893             return None
    894 
--> 895         return pybamm.lithium_ion.ElectrodeSOHSolver(
    896             self.parameter_values, self.model.param
    897         )

/usr/local/lib/python3.8/dist-packages/pybamm/models/full_battery_models/lithium_ion/electrode_soh.py in __init__(self, parameter_values, param, known_value)
    166         # Check whether each electrode OCP is a function (False) or data (True)
    167         OCPp_data = isinstance(parameter_values["Positive electrode OCP [V]"], tuple)
--> 168         OCPn_data = isinstance(parameter_values["Negative electrode OCP [V]"], tuple)
    169 
    170         # Calculate stoich limits for the open circuit potentials

/usr/local/lib/python3.8/dist-packages/pybamm/parameters/parameter_values.py in __getitem__(self, key)
    137 
    138     def __getitem__(self, key):
--> 139         return self._dict_items[key]
    140 
    141     def get(self, key, default=None):

/usr/local/lib/python3.8/dist-packages/pybamm/util.py in __getitem__(self, key)
     66                 )
     67             best_matches = self.get_best_matches(key)
---> 68             raise KeyError(f"'{key}' not found. Best matches are {best_matches}")
     69 
     70     def search(self, key, print_values=False):

KeyError: "'Negative electrode OCP [V]' not found. Best matches are ['Primary: Negative electrode OCP [V]', 'Positive electrode OCP [V]', 'Negative electrode thickness [m]']"
```

tinosulzer, 2023-02-13:
If you wanted to calculate the capacity, you would need to implement this. Which would be a useful and welcome contribution 
[image]

BradyPlanden, 2023-02-14:
Thanks, Tino. Setting `calc_esoh=False` fixed the error. Would it be worthwhile to include a cycling example for the two-phase implementation with this `calc_esoh=False`?

That makes sense. I'll start down the road of implementing the above; am I correct in thinking that this implementation should sit inside `electrode_soh.py`? I suspect we could use `"particle phases"` as a condition for the above to be utilised; however, maybe there is an alternative that would be more suitable.

tinosulzer, 2023-02-15:
Yes, it would be worthwhile to add a cycling example to the end of the existing two-phase example. And you are correct in your thinking for the second part

KennethNwanoro, 2023-02-15:
Hi all,

I have similar problem running Chen2020_composite parameter set with pybamm experiment. Below is my example script.

```
import pybamm

options = {
    "particle phases": ("2", "1"),
    "open circuit potential": (("single", "current sigmoid"), "single"),
}

model = pybamm.lithium_ion.DFN(options)
parameter_values = pybamm.ParameterValues("Chen2020_composite")

experiment =  pybamm.Experiment([
         ("Discharge at C/20 for 20 hours or until 2.6 V",),
     ]*1)

sim = pybamm.Simulation(model, experiment=experiment, solver = pybamm.CasadiSolver())
sol = sim.solve(calc_esoh=False)
```

KennethNwanoro, 2023-02-15:
Setting `calc_esoh = False` did not resolve the issue.

KennethNwanoro, 2023-02-15:
Here is my traceback:

```
Traceback (most recent call last):

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 639, in process_symbol
    return self._processed_symbols[symbol]

KeyError: Division(-0x7658b1bbeffa1e35, /, children=['96485.33212 * Primary: Maximum concentration in negative electrode [mol.m-3] * (Negative electrode thickness [m] + Separator thickness [m] + Positive electrode thickness [m])', 'absolute(Typical current [A] / (Number of electrodes connected in parallel to make a cell * Electrode width [m] * Electrode height [m]))'], domains={})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 639, in process_symbol
    return self._processed_symbols[symbol]

KeyError: Multiplication(-0x77e385336f4b5b24, *, children=['96485.33212 * Primary: Maximum concentration in negative electrode [mol.m-3]', 'Negative electrode thickness [m] + Separator thickness [m] + Positive electrode thickness [m]'], domains={})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 639, in process_symbol
    return self._processed_symbols[symbol]

KeyError: Multiplication(-0x2ca91095ccd27d27, *, children=['96485.33212', 'Primary: Maximum concentration in negative electrode [mol.m-3]'], domains={})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 639, in process_symbol
    return self._processed_symbols[symbol]

KeyError: Parameter(0x7c1355ce48b0c9f9, Primary: Maximum concentration in negative electrode [mol.m-3], children=[], domains={})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "lib\site-packages\pybamm\util.py", line 58, in __getitem__
    return super().__getitem__(key)

KeyError: 'Primary: Maximum concentration in negative electrode [mol.m-3]'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "untitled1.py", line 23, in <module>
    sol = sim.solve(calc_esoh=False)

  File "lib\site-packages\pybamm\simulation.py", line 612, in solve
    self.build_for_experiment(check_model=check_model, initial_soc=initial_soc)

  File "lib\site-packages\pybamm\simulation.py", line 447, in build_for_experiment
    self.set_up_and_parameterise_experiment()

  File "lib\site-packages\pybamm\simulation.py", line 144, in set_up_and_parameterise_experiment
    timescale = self._parameter_values.evaluate(model.timescale)

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 811, in evaluate
    processed_symbol = self.process_symbol(symbol)

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 641, in process_symbol
    processed_symbol = self._process_symbol(symbol)

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 752, in _process_symbol
    new_left = self.process_symbol(symbol.left)

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 641, in process_symbol
    processed_symbol = self._process_symbol(symbol)

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 752, in _process_symbol
    new_left = self.process_symbol(symbol.left)

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 641, in process_symbol
    processed_symbol = self._process_symbol(symbol)

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 753, in _process_symbol
    new_right = self.process_symbol(symbol.right)

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 641, in process_symbol
    processed_symbol = self._process_symbol(symbol)

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 650, in _process_symbol
    value = self[symbol.name]

  File "lib\site-packages\pybamm\parameters\parameter_values.py", line 139, in __getitem__
    return self._dict_items[key]

  File "lib\site-packages\pybamm\util.py", line 68, in __getitem__
    raise KeyError(f"'{key}' not found. Best matches are {best_matches}")

KeyError: "'Primary: Maximum concentration in negative electrode [mol.m-3]' not found. Best matches are ['Maximum concentration in negative electrode [mol.m-3]', 'Maximum concentration in positive electrode [mol.m-3]', 'Initial concentration in negative electrode [mol.m-3]']"
```

BradyPlanden, 2023-02-15:
Add `parameter_values = parameter_values` into `pybamm.Simulation`.

KennethNwanoro, 2023-02-15:
Thanks a lot, I missed that part after trying to resolve the issue previously raised.

rtimms, 2025-02-10:
cc @aabills