pybamm-team/PyBaMM #2683: Issue with using NCA_Kim2011 chemistry

zafar594, 2023-02-13:
I'm getting error "**must provide 'cell' parameters for NCA_Kim2011 chemistry**"
What does this error mean exactly?
Also, this error does not occur when using Chen2020, Xu2019 or Prada2013 chemistries in same program, so I think my program is working fine.

tinosulzer, 2023-02-13:
Can you post your full script and which pybamm version you are using please

zafar594, 2023-02-14:
The entire program is too long to post here so I am just posting the segment where if-else loop is selecting different battery chemistries, among Chen, Xu, Prada and Kim.

**elif positive_electrode_chemistry == 'NCA':
        chemistry = {"chemistry": pybamm.parameter_sets.NCA_Kim2011}
        for key,values in pybamm.ParameterValues(chemistry=chemistry).items():
            if key in str(parameter_values):
                parameter_values[key] = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.NCA_Kim2011)[key]
            else:
                parameter_values.update({key:values}, check_already_exists = False)**

This is returning the error:
[image]

I am using same code for other chemistries and the program runs fine (ie, when I remove Kim2011 from the program)
I'm using Python 3.9.7 and Pybamm 22.12 (22.11 also gives same result).
Thanks!

tinosulzer, 2023-02-14:
Try just doing `pybamm.ParameterValues("NCA_Kim2011")`

zafar594, 2023-02-14:
Thanks! This worked!

closed 2023-02-14 by zafar594