pybamm-team/PyBaMM #2689: I wrote a program to convert power to current at every time step using simple ohm's law for a battery pack with resistances in terminals and busbar with PyBaMM. I simulate just a single time step between each  update, but no matter how I change the drive cycle (averaging two time samples, omitting alternate samples, etc.), it seems that the process never proceeds further after "Start solver set-up" or while descritization. I have to terminate it manually.

cyasing, 2023-02-14:
I wrote a program to convert power to current at every time step using simple ohm's law for a battery pack with resistances in terminals and busbar with PyBaMM. I simulate just a single time step between each  update, but no matter how I change the drive cycle (averaging two time samples, omitting alternate samples, etc.), it seems that the process never proceeds further after "Start solver set-up" or while descritization. I have to terminate it manually.

What can be the possible reasons? I monitored some parameters like current, voltage, power, temperature, sei length, and concentration for each cell and time step, but nothing seems out of order.

Discretisation error:

[image]

Solver error:

[image]
[image]

_Originally posted by @cyasing in https://github.com/pybamm-team/PyBaMM/discussions/2688_

cyasing, 2023-02-14:
I wrote a program to convert power to current at every time step using simple ohm's law for a battery pack with resistances in terminals and busbar with PyBaMM. I simulate just a single time step between each  update, but no matter how I change the drive cycle (averaging two time samples, omitting alternate samples, etc.), it seems that the process never proceeds further after "Start solver set-up" or while descritization. I have to terminate it manually.

cyasing, 2023-02-15:
Can I instead send the whole script? Only if you're okay with it.

tinosulzer, 2023-02-15:
You can, but it decreases the chance of your issue being fixed. See https://stackoverflow.com/help/minimal-reproducible-example

cyasing, 2023-02-16:
Right. I did it in two different ways: 
1. mrc_half.py uses a research paper's equations (based on ohm's law)
2. (mrc_half_net.py) uses two functions of netlist_utils in liionpack for resolving the array of cells (did not import liionpack), then proceeded as normal.

Both end up stopping at different points (91-92 and 56-57 respectively), somehow both crossed the 42-43 mark. I am attaching them both here, along with the data:
[mrcs.zip](https://github.com/pybamm-team/PyBaMM/files/10754974/mrcs.zip)

cyasing, 2023-02-21:
@tinosulzer Anything of note that you found?

closed 2023-02-21 by cyasing