pybamm-team/PyBaMM #2691: Add a model for symmetric Li/Li cells (2 planar electrodes plus homogenous electrolyte)

RuiheLi, 2023-02-14:
### Description

Add a model for symmetric Li/Li cells (2 planar electrodes plus homogenous electrolyte)

### Motivation

Provide another common model in literatures especially for basic electrolyte characterization. 
Eventually become a PyBaMM maintainer by taking this task.

MarcAntoine88, 2023-02-15:
Hello @RuiheLi I've done a first version of Li vs Li model but it is not yet dimensionless, I still need to work on it.

brosaplanella, 2023-02-15:
This feature should be fairly easy to implement. Maybe a way to proceed is the following

- [ ] Write an example where you assemble the model from existing submodels there, using the lithium-metal submodels for the electrode.
- [ ] Validate that model to ensure it gives the right behaviour.
- [ ] Define a new model class for the symmetric cell. 

There are some extra things that could be done, like define the electrolyte to be homogeneous instead of porous, so porosity and tortuosity are not required parameters.

jaskiratsingh2000, 2023-02-20:
Hi! Do we have any description for the model? On requirements or formulas?