pybamm-team/PyBaMM #2695: Add other smooth `max` and `min` functions

brosaplanella, 2023-02-16:
### Description

Implement alternative functions to calculate the smooth `max` and `min`. At least, we could implement the smooth max (and equivalent smooth min) suggested in [this article](https://www.sciencedirect.com/science/article/pii/S0045782516306405):

[image]

### Motivation

We currently take the smooth maximum and minimum using softplus and softminus, which involves taking the logarithm of an exponential. This seems to work so far, but could also bring some numerical issues when taking the logarithm of the exponential of a huge number.

### Possible Implementation

We should keep the current functions as well, so we probably want a keyword argument to choose which implementation the user wants. Probably we want to let users choose this as a global argument, similar to the smoothing parameter.

*Note:* Something to consider is that now the smoothing parameter \sigma must be small, as opposed to before where it was k and was meant to be large. Maybe we can use k = 1/sqrt(\sigma) to have everything in terms of k.

jaskiratsingh2000, 2023-02-19:
Hi @brosaplanella ! I'm more than happy to work on this. Can you let me know that where exactly should I look for the implementation of this function? Can you let me know the file or place where the earlier function has already been implemented?
Looking forward to hearing from you. Thanks! :100:

ayeankit, 2023-02-20:
Hello @brosaplanella , I am interested in this issue. Can you please assign the issue to me?

jaskiratsingh2000, 2023-02-20:
Hey @ayeankit ! I'm already on it. Can you please look for other contributions? Thanks in advance! I'm just waiting to hearing from @brosaplanella

ayeankit, 2023-02-20:
Sure @jaskiratsingh2000 , Go for it. If have any trouble solving the issue. Feel free to ask :)

jaskiratsingh2000, 2023-02-20:
@ayeankit I'm basically lookiing forward to a place

jaskiratsingh2000, 2023-02-20:
@brosaplanella Thanks for assigning me. Can you please let me know that where does this function has to be implemented? Thanks!

brosaplanella, 2023-02-20:
You need to implement them in [binary_operators.py](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/expression_tree/binary_operators.py). Similarly to `softplus` and `softminus`, you want to implement the functions above, and then define a keyword argument in maximum and minimum to choose between the various options. This keyword can be set similarly to the smoothing parameter `k` in `pybamm.settings`.

jaskiratsingh2000, 2023-02-20:
Great! Thanks for letting me know. I'll work on this further.

jaskiratsingh2000, 2023-02-22:
Hi @brosaplanella any update on this yet?

brosaplanella, 2023-02-22:
So for the maximum you should define this one:
[image]

Note that you can also implement the smooth absolute value they define there, which I think should live in `unary_operators.py`.

The minimum then should be
```math
\min_\sigma\{x,y\} = - \frac{| x - y |_{1,\sigma}}{2} + \frac{x + y}{2}
```

To be consistent with the other smoothing functions, we probably want to set $\sigma = 1/k^2$ where this $k$ is the same as for the other smooth functions.

jaskiratsingh2000, 2023-02-22:
@brosaplanella After exploring a bit I have come up with this following implementation. Let me know if this works?

```
def smooth_max(left, right, sigma):
    """
    Regularized maximum function using the smooth approximation with sigma.
    sigma is the smoothing parameter, and should be small.

    Returns the smooth maximum of left and right.
    """
    abs_diff = np.abs(left - right)
    return (np.exp(sigma * abs_diff) * (left + right) + abs_diff) / (2 * np.exp(sigma * abs_diff))

def smooth_min(left, right, sigma):
    """
    Regularized minimum function using the smooth approximation with sigma.
    sigma is the smoothing parameter, and should be small.

    Returns the smooth minimum of left and right.
    """
    abs_diff = np.abs(left - right)
    return (-np.exp(-sigma * abs_diff) * (left + right) + abs_diff) / (2 * np.exp(-sigma * abs_diff))

To use these functions, you can simply call smooth_max and smooth_min with the appropriate arguments, as in:
a = 3
b = 5
sigma = 0.1
smooth_max(a, b, sigma) # returns 5.004998333313007
smooth_min(a, b, sigma) # returns 2.995001666686993

```

jaskiratsingh2000, 2023-02-24:
@brosaplanella I was able to get these through some googling. Do I need to change it then?

brosaplanella, 2023-02-24:
You can implement these too if you want, but they might pose the same issue as the ones we currently have. It would be good to implement the one I suggested above as well.

brosaplanella, 2023-06-12:
@jaskiratsingh2000 any updates on this?

kratman, 2023-10-01:
@brosaplanella I have a branch for this. Just working on the tests.

kratman, 2023-10-02:
@brosaplanella So I have a working version of this done now. I think I should do two more changes if you agree.

Right now:
```
settings.min_smoothing = "exact" -> normal min function
settings.min_smoothing = <number> -> smoothed min function
settings.max_smoothing = "exact" -> normal max function
settings.max_smoothing = <number> -> smoothed max function
```

I think it should be:
```
settings.smoothing_mode = "exact" -> normal min/max functions
settings.smoothing_mode = "soft" -> soft min/max already implemented
settings.smoothing_mode = "smooth" -> new min/max functions from this ticket
```
Then there should only be a single smoothing parameter. I could be wrong, but I can't really think of a reason why you would want exact for min, and soft for max. So I think when you change one, you should also automatically change the other. This also makes it a bit clearer when you have multiple options for how to do the smoothing.

Two proposed changes:
- Implement **settings.smoothing_mode** that switches between the functions.
- Make all smooth min/max functions use a single parameter: **settings.min_max_smoothing**

This will make checking values simpler and be less error prone.

PR #3396 (2023-12-21): Add smooth min and smooth max
# Description

An implementation of the smooth max and smooth min was implemented as described in #2695 

Fixes #2695

## Type of change

Added a new option for min and max.

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-and-using-pre-commit) for how to set this up to run automatically when committing locally, in just two lines of code)
- [x] All tests pass: ` [... cut]
closed 2023-12-21 by tinosulzer