pybamm-team/PyBaMM #2725: Reduce testing on Windows and Mac

tinosulzer, 2023-02-23:
I'm wondering if there's any reason to run the integration tests on Windows and Mac. The unit tests should be sufficient (and maybe even for Linux python 3.8). If we then moved the integration and examples testing on 3.9 only to new processes in parallel, the whole test suite would be much faster.

brosaplanella, 2023-02-23:
That's a good point. Is it the reasoning that given that the code is the same across platforms, if the building blocks work in all platforms (unit tests) so should do when assembled together (integration tests)?

Integration tests also tie more nicely with validation and part of the benchmarking, so might be useful to have them separately. What do other packages do.

tinosulzer, 2023-02-23:
> Is it the reasoning that given that the code is the same across platforms, if the building blocks work in all platforms (unit tests) so should do when assembled together (integration tests)?

yes

> Integration tests also tie more nicely with validation and part of the benchmarking, so might be useful to have them separately.

It's still useful to have them running on every PR since often new code can pass the unit tests but still break integration tests

martinjrobins, 2023-02-23:
seems reasonable, but I would at least run the integration tests before publishing a new release, just to be sure you arn't introducing any mac/windows specific bugs for users.

tinosulzer, 2023-04-05:
So integration tests (python 3.8 on windows and mac) should be run when opening a PR to main with option to triggered manually?

agriyakhetarpal, 2023-04-05:
Perhaps integration tests can also be skipped to run on forks too, since that would be redundant when opening a PR to main

tinosulzer, 2023-04-05:
Yes. Another thought is that we can run the integration tests as part of the nightly run. So to clarify:
- run all tests on linux with the latest python
- run unit tests only on other OS and python versions
- run all tests on all OS and python versions nightly and for PRs to main

Behavior on PRs from forks and branches should be identical

closed 2023-05-18 by tinosulzer