pybamm-team/PyBaMM #2726: Rename `one_plus_dlnf_dlnc` to `thermodynamic_factor`

tinosulzer, 2023-02-23:
Maybe for a separate (small) issue, should we rename this to `thermodynamic_factor`? It is not much longer, and more informative.

_Originally posted by @brosaplanella in https://github.com/pybamm-team/PyBaMM/pull/2419#discussion_r1116150328_

tinosulzer, 2023-02-23:
Changes to be made:
- everywhere `one_plus_dlnf_dlnc` appears, replace it with `thermodynamic_factor`
- everywhere "1 + dlnf/dlnc" appears, replace it with "Thermodynamic factor"
- add specific check and deprecation message [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/parameters/parameter_values.py#L297) if `values = "1 + dlnf/dlnc"`

ayeankit, 2023-02-24:
Hello @tinosulzer , Can I work this issue?

jeromtom, 2023-02-24:
I am working on this issue with my PR #2727, and it might take some more time. I would like this issue to be assigned to me. Thank you.

PR #2727 (2023-03-12): Rename one_plus_dlnf_dlnc to thermodynamic_factor
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #2726

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [X] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

 [... cut]
closed 2023-03-12 by tinosulzer