pybamm-team/PyBaMM #2730: [Bug]: Half cell model error when working electrode assigned to 'negative electrode'. Will not run

ss711, 2023-02-24:
### PyBaMM Version

23.1

### Python Version

3.9.13

### Describe the bug

When trying to use the half cell DFN model for the negative electrode, specifying in 'options' that the 'working electrode' is the 'negative electrode' results in an error just saying 'positive electrode'. However, there is no error when 'working electrode' is the 'positive electrode'. Same error when using both simulation and mesh solver approach.

### Steps to Reproduce

import pybamm

parameter_values = pybamm.ParameterValues("OKane2022")

Crates = [0.5] #,1,2,5,6,7,8,9]

for i in Crates:

    options = {"working electrode" : "negative"}
    model = pybamm.lithium_ion.DFN(options=options)
    sim = pybamm.Simulation(model, parameter_values=parameter_values, solver=pybamm.CasadiSolver(),C_rate = i)
    sim.solve([0, 36000])
    solution = sim.solution 

### Relevant log output

```shell
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Cell In[10], line 18
     15 parameter_values["Current function [A]"] = i * parameter_values['Nominal cell capacity [A.h]']   # max cap = 5Ah 
     17 options = {"working electrode" : "negative"}
---> 18 model = pybamm.lithium_ion.DFN(options=options)
     19 geometry = model.default_geometry
     20 parameter_values.process_model(model)

File ~\Documents\Python\env\lib\site-packages\pybamm\models\full_battery_models\lithium_ion\dfn.py:35, in DFN.__init__(self, options, name, build)
     32 self.x_average = False
     33 super().__init__(options, name)
---> 35 self.set_submodels(build)
     37 pybamm.citations.register("Doyle1993")

File ~\Documents\Python\env\lib\site-packages\pybamm\models\full_battery_models\lithium_ion\base_lithium_ion_model.py:91, in BaseModel.set_submodels(self, build)
     88 self.set_total_interface_submodel()
     90 if build:
---> 91     self.build_model()

File ~\Documents\Python\env\lib\site-packages\pybamm\models\full_battery_models\base_battery_model.py:953, in BaseBatteryModel.build_model(self)
    950 def build_model(self):
    951 
    952     # Build model variables and equations
--> 953     self._build_model()
    955     # Set battery specific variables
    956     pybamm.logger.debug("Setting voltage variables ({})".format(self.name))

File ~\Documents\Python\env\lib\site-packages\pybamm\models\base_model.py:553, in BaseModel._build_model(self)
    550 pybamm.logger.info("Start building {}".format(self.name))
    552 if self._built_fundamental is False:
--> 553     self.build_fundamental()
    555 self.build_coupled_variables()
    557 self.build_model_equations()

File ~\Documents\Python\env\lib\site-packages\pybamm\models\base_model.py:454, in BaseModel.build_fundamental(self)
    448 for submodel_name, submodel in self.submodels.items():
    449     pybamm.logger.debug(
    450         "Getting fundamental variables for {} submodel ({})".format(
    451             submodel_name, self.name
    452         )
    453     )
--> 454     self.variables.update(submodel.get_fundamental_variables())
    456 self._built_fundamental = True

File ~\Documents\Python\env\lib\site-packages\pybamm\models\submodels\electrolyte_conductivity\full_conductivity.py:38, in Full.get_fundamental_variables(self)
     35     phi_e_k.print_name = f"phi_e_{domain[0]}"
     36     phi_e_dict[domain] = phi_e_k
---> 38 variables = self._get_standard_potential_variables(phi_e_dict)
     39 return variables

File ~\Documents\Python\env\lib\site-packages\pybamm\models\submodels\electrolyte_conductivity\base_electrolyte_conductivity.py:55, in BaseElectrolyteConductivity._get_standard_potential_variables(self, phi_e_dict)
     51     phi_e_s = phi_e_dict["separator"]
     52     phi_e_dict["negative electrode"] = pybamm.boundary_value(phi_e_s, "left")
     54 eta_e_av = pybamm.x_average(
---> 55     phi_e_dict["positive electrode"]
     56 ) - pybamm.x_average(phi_e_dict["negative electrode"])
     57 phi_e_av = pybamm.x_average(phi_e)
     59 variables = {
     60     "Electrolyte potential": phi_e,
     61     "Electrolyte potential [V]": -U_ref + pot_scale * phi_e,
   (...)
     66     "Gradient of electrolyte potential": pybamm.grad(phi_e),
     67 }

KeyError: 'positive electrode'
```

tinosulzer, 2023-05-15:
To fix this: add better error message when "working electrode": "negative", and implement parameter-flipping utility function

rtimms, 2023-05-17:
Question: How best to handle adding degradation to a half-cell, since all the degradation mechanisms right now only work for "negative electrode"?

DrSOKane, 2023-05-17:
There are two levels of sophistication here.

The quick fix is to add an if statement that multiplies the terminal voltage and related variables by -1 if "working electrode" is set to "negative", so that the energy stored by the half-cell is positive.

The more rigorous method is to set graphite to be the positive electrode. This does indeed raise issues with degradation, but there is a strong case for implementing CEI (aka pSEI) on the positive electrode as many rival software packages already do so. There is also no _a priori_ reason not to enable lithium plating on the positive electrode, except for numerical difficulties because the amount of plated lithium will always be close to zero for most positive electrode materials.

rtimms, 2023-05-18:
OK, so one option would be to:

- Remove the "working electrode" option and instead have an option that sets up half-cell or full-cell. In a full-cell the domains are always "negative electrode", separator", "positive electrode". In a half-cell the domains are always "separator", "positive electrode".
- Make the parameter flipping (or extraction) utility that gets the "negative" or "positive" electrode (and current collector?) parameters from a full-cell parameter set and returns a dict with them all labelled "positive" for use with a half-cell model. Then combine this with parameters for Li-metal counter electrode. Maybe we should have a standard Li-metal half-cell parameter set (e.g. with the geometry set up for a coin cell) and utility that lets you slot in positive electrode parameters, with a nominal half-cell capacity calculated based on the thickness, c_max, active material volume fraction, etc.
- Make the degradation models that are currently only implemented for "negative electrode" be compatible with either "negative electrode" or "positive electrode" and pass in a 2-tuple for the options. My only worry with this is what should be the expected behaviour of providing a single value for the option in a full-cell? E.g. should `{"sei": "reaction limited"}` follow existing behaviour and set the SEI model to reaction limited in both electrodes, or would people expect that  `{"sei": "reaction limited"}` would just set the model for the negative electrode, as is more common? 

@DrSOKane I'm not super familiar with CEI - is the model/mechanism fundamentally different or do people just give a different name to the same model applied in a different electrode?

DrSOKane, 2023-05-19:
pSEI is a different mechanism, caused by transition metal ions dissolving into the electrolyte and forming oxides or fluorides that precipitate on the electrode surface. Most papers I've seen use a simple Tafel equation to describe it, but some e.g. Xianke Lin et al. (2013) propose a much more rigorous mechanism mediated by protons. Regardless of the mechanism, the default behaviour should be to have no (p)SEI on the positive electrode unless a tuple is provided.

tinosulzer, 2023-05-19:
It probably makes sense to just have a different option for "SEI" and "pSEI" (or "CEI"?), rather than a tuple.

brosaplanella, 2023-05-19:
> It probably makes sense to just have a different option for "SEI" and "pSEI" (or "CEI"?), rather than a tuple.

I think this is a good idea, at least for now that we don't know much about pSEI/CEI. Though I guess this ties back to the original problem. For example, we could do that specifying SEI models for positive working electrode throws a warning saying the option was ignore, but that does not transfer nicely to having a "negative electrode half cell".

Having said that, I think half-cells for testing purposes are not too stable, so maybe it is not an issue anyway...

rtimms, 2023-05-19:
I agree it makes sense for separate SEI/CEI options. My thinking for allowing a tuple for SEI was so that you could have a half-cell where the "positive electrode" is graphite and you might like to include SEI.

DrSOKane, 2023-06-05:
Hi all, I'm currently working on this but unsure what to call the new variables! Would you prefer "Inner negative SEI thickness [m]" or "Negative inner SEI thickness [m]"?

PR #3198 (2023-10-03): Negative half-cell models
# Description

This pull request introduced negative half-cell models to PyBaMM, by means of three major changes:
* The option `"working electrode": ["both", "negative", "positive"]` has been replaced with `"half-cell": ["false", "true]`. The positive electrode is always considered to be the working electrode, which has always been the case but is now made explicit.
* Two new parameter sets - `Ecker2015_graphite_halfcell` and `OKane2022_negative_halfcell` - have been added to work with the new option.
* The degradation mechanisms `SEI`, `SEI on cracks` and `lithium pl [... cut]
closed 2023-10-03 by brosaplanella