pybamm-team/PyBaMM #2757: ExperimentalData class

brosaplanella, 2023-03-09:
### Description

While PyBaMM is a modeling package, the majority of battery research is performed through experiments, whose data (cell voltage, current, temperature, ...) can be generated in a wide variety of formats. The goal of this project is to develop functionality to better interface PyBaMM with experimental data. In particular, the proposed ExperimentalData class should import real data and behave like the existing Solution class (generated by solving a model), so experimental data can be easily plotted and compared with simulations.

### Motivation

Facilitate comparisons between models and experimental data.

martinjrobins, 2023-03-09:
# Hackathon info
This will be one of the projects for the first PyBaMM Hackathon. Here you can find some information (summarising other points made on this issue) which should be a useful starting point. Note that this post will evolve, even during the Hackathon day, so make sure to check it regularly.

## Background

See description above

## Tasks
- [ ] Implement `ExperimentalData` class with similar public facing API as `Solution`
- [ ] Write tests for `ExperimentalData` against expected behavior
- [ ] Implement CSV data import capability (other formats, e.g. biologic/maccor/arbin...?) and tests
- [ ] Implement plotting capability so can be compared with `Solution`
- [ ] Write example notebook, illustrating plotting and comparison (e.g. RMS error) with solutions
- [ ] Ensure coverage and tests pass.

jayceslesar, 2023-03-14:
Might be out of scope but might be very interesting to be able to use this repo as a common interface into other battery test equipment....say you could create an experiment here and have methods on the experiment instance that create the required input for real test equipment? see https://github.com/jepegit/cellpy for a repo containing common hardware to do this