pybamm-team/PyBaMM #2766: [Bug]: readthedocs version switcher does not style properly for older PyBaMM versions

agriyakhetarpal, 2023-03-12:
### PyBaMM Version

23.2

### Python Version

3.9.13

### Describe the bug

1. The [version switcher dropdown](https://pydata-sphinx-theme.readthedocs.io/en/v0.8.1/user_guide/configuring.html?highlight=orange#style-the-switcher-buttons) does not style properly on [readthedocs]( https://pybamm.readthedocs.io): version text is displayed in a green coloured font regardless of the active version (which should ideally be `orange` for dev, `green` for the latest version, and `red` for older versions of PyBaMM)

Some other related issues: 

2. Links with the version switcher to the API documentation for past versions are broken (though this is because the API documentation index page was earlier not hosted separately)
3. The documentation does not use the PyData Theme for PyBaMM versions 22.12 and older, and uses the old theme instead

### Steps to Reproduce

Hosted on readthedocs and with the [instructions to build docs](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#building-the-documentation)

tinosulzer, 2023-03-12:
Can you open a PR to fix point 1? For points 2 and 3, they're past versions, I don't think we need to change it

agriyakhetarpal, 2023-03-12:
Sure, perhaps I can open a PR for point 1 – otherwise please feel free to close this

tinosulzer, 2023-03-12:
Yes, point 1 is worth fixing. Also add the version switcher to the header top right if you can

PR #2769 (2023-03-13): Fix version switcher on readthedocs
# Description

This updates the version switcher styling on readthedocs and configures a `.json` file for the version switcher dropdown

Fixes #2766 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (no [... cut]
closed 2023-03-13 by tinosulzer