pybamm-team/PyBaMM #2788: [Bug]: Solution `last_state` can't be used as a starting solution

brosaplanella, 2023-03-20:
### PyBaMM Version

23.2

### Python Version

3.9.16

### Describe the bug

When trying to use the `last_state` as a starting solution for a new simulation it breaks because
```
AttributeError: 'Solution' object has no attribute 'steps'
```

The issue seems to be here: https://github.com/pybamm-team/PyBaMM/blob/81d8c351579af9ed1196805eab8e36ca61ca2468/pybamm/simulation.py#L663

### Steps to Reproduce

```python3
import pybamm

model = pybamm.lithium_ion.SPM()

experiment = pybamm.Experiment(["Discharge at 1C until 3.5 V"])
sim = pybamm.Simulation(model, experiment=experiment)
solution = sim.solve()

experiment2 = pybamm.Experiment(["Discharge at 1C until 2.5 V"])
sim2 = pybamm.Simulation(model, experiment=experiment2)
solution2 = sim2.solve(starting_solution=solution.last_state)
```

### Relevant log output

```shell
Traceback (most recent call last):
  File "/home/brosaplanella/support-pybamm/.github/profile/mwe.py", line 11, in <module>
    solution2 = sim2.solve(starting_solution=solution.last_state)
  File "/home/brosaplanella/support-pybamm/env/lib/python3.9/site-packages/pybamm/simulation.py", line 663, in solve
    starting_solution.steps,
AttributeError: 'Solution' object has no attribute 'steps'
```

tinosulzer, 2023-03-20:
This is a bug, easiest way to fix it is to add some extra logic for that case in `simulation.py` (i.e. initialize steps to be an empty list)

manjunathnilugal, 2023-03-22:
Is there is any method to fix this issue as of now until the issues is resolved.
It is mentioned to to add `steps = [ ]`, in simulation.py file, @tinosulzer could you please tell in brief. 
It would be very helpful.

manjunathnilugal, 2023-03-29:
How to implement this in the current version of  pybamm, **i.e. in my local system**

jsbrittain, 2023-03-29:
@manjunathnilugal Hi - The change has been merged so should be available immediately in the `develop` branch. If you want to make this change in your local system, you will need to [alter this line in simulation.py](https://github.com/pybamm-team/PyBaMM/blob/c78bed2a985c3ff46cdc2e54d20e62736c2e83b9/pybamm/simulation.py#L655) to `[starting_solution]` [as here](https://github.com/pybamm-team/PyBaMM/blob/8232ac4e14b072e4f5f4316342e7953adef70c9d/pybamm/simulation.py#L655).

PR #2822 (2023-03-29): Fix use of last-state as starting-solution in Simulation.solve()
# Description

When a starting_solution was provided to Solution.solve(), the method attempted to pass a sub-field of the starting_solution to make_cycle_solution(), instead of the required list of Solver objects.

Fixes #2788 ([Bug]: Solution last_state can't be used as a starting solution)

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to th [... cut]