pybamm-team/PyBaMM #2789: [Bug]: Short discharge pulse in drive cycle partially ignored

DavidMStraub, 2023-03-20:
### PyBaMM Version

v23.2

### Python Version

3.8.10

### Describe the bug

I am simulating a drive cycle with a 1C discharge for a couple of minutes, followed by a rest phase, followed by a 10 second 1C discharge (beginning of a typical HPPC test). I am using a drive cycle rather than an Experiment as I am fitting to experimental data.

I noticed that, during the short pulse, the predicted voltage is not as expected and traced it back to the absence of any diffusive contribution; in fact the concentration in the electrodes does not change during the pulse. So it looks like part of the simulation (diffusion equation) ignores the presence of the pulse.

### Steps to Reproduce

MWE:

```python
import matplotlib.pyplot as plt
import numpy as np
import pybamm

model = pybamm.lithium_ion.SPM()
Q = model.default_parameter_values["Nominal cell capacity [A.h]"]

time = np.arange(0, 1000, 2, float)
current = 0 * time
current[(time <= 500)] = Q
current[(time >= 800) & (time <= 810)] = Q

param = model.default_parameter_values.copy()
param["Current function [A]"] = pybamm.Interpolant(
    time, current, model.timescale * pybamm.t
)

sim = pybamm.Simulation(model, parameter_values=param)
sol = sim.solve()
```

Plot:

```python
plt.plot(time, current)
plt.plot(sol["Time [s]"].entries, sol["Current [A]"].entries, ls="--")
plt.show()

for domain in ["positive", "negative"]:
    plt.plot(
        sol["Time [s]"].entries, sol[f"Average {domain} particle concentration"].entries
    )
plt.show()
```

Output: 

[image]

So while the current correctly shows the short peak, it has no impact on concentration.

Note that by playing with the length of the discharge and rest phases, one can make the issue appear and disappear.

martinjrobins, 2023-03-22:
Hi @DavidMStraub. I'm guessing that the solver is stepping over these discontinuities in your current function interpolant. We try to detect discontinuities in the equations given to a model, but we don't try and detect discontinuities in interpolants. You can manually add discontinuity events to your model like so:

```python
model.events.append(pybamm.Event("current_disc1", 500 / model.timescale , pybamm.EventType.DISCONTINUITY))
model.events.append(pybamm.Event("current_disc2", 1000 / model.timescale, pybamm.EventType.DISCONTINUITY))
model.events.append(pybamm.Event("current_disc3", 1010 / model.timescale, pybamm.EventType.DISCONTINUITY))
```

Do this before passing your `model` to the `pybamm.Simulation` contructor. Let me know if this fixes your problem.

martinjrobins, 2023-03-24:
This looks like a bug in the fast casadi solver, I can have a look at this and try to work out what is going on. In the meantime, have you tried the IDAKLU solver, or the "slow" CasadiSolver? Both of these treat events and sensitivities differently, so you might have more success with these

DavidMStraub, 2023-03-24:
Thanks!

Actually Using `mode="safe"` in the last example does not solve the error, same for `"fast with events"`. I have not tried out IDAKLU yet but this might be a good opportunity. I need to follow [these instructions](https://pybamm.readthedocs.io/en/v0.2.3/install/install-klu.html), correct?

DavidMStraub, 2023-03-24:
Sorry, I realized I was on a developer install, but installing the wheel `IDAKLUSolver` works out of the box.

DavidMStraub, 2023-03-24:
So with IDAKLU the above example works (the step in concentration is present) even without manually adding the events.

I do run into another (unrelated) issue though: calling

```python
sol["Terminal voltage [V]"].sensitivities
```
raises a `KeyError` because the `"all"` key is missing, but actually

```python
sol["Terminal voltage [V]"].solution_sensitivities["Negative particle radius [m]"]
```
does work...

DavidMStraub, 2023-03-24:
(I should note that my IDAKLU experiments are done with PyBaMM v22.12 rather than v23.2.

martinjrobins, 2023-03-24:
two bugs to fix then ;) But I'm glad the IDAKLU solver is workable for you, the way it treats events is much superior to the casadi solver. Eventually we're likely to make the idaklu solver the default pybamm solver.

PR #2883 (2023-04-19): #2789 add "all" to sensitivities for idaklu solver
# Description

Partially fixes #2789 and unblocks #2858 

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-and-using-pre-commit) for how to set this up to run automatically when committing locally, in just two lines of code)
- [x] All tests pass: `$ python run-tests.py --all`
- [x] The documentation builds: `$ python run-tests.py --doctest`

You can run unit [... cut]
closed 2023-04-19 by tinosulzer