pybamm-team/PyBaMM #2796: New solver mode: only including required variables (don't store entire `y`)

martinjrobins, 2023-03-22:
### Description

Provide a mode for the solver so that only a user given list of variables are calculated and stored as part of the solution

### Motivation

Remove the need to store the entire `y` solution and store only the variables given by the user

### Possible Implementation

Interface might look like the following:

```python
solver = pybamm.IDAKLUSolver()
sol = solver.solve(model, t_eval, solution_include=['Voltage [V]'])
print(sol['Voltage [V]'].entries)

sol['Negative particle surface concentration']  # gives an exception
```

tinosulzer, 2023-03-24:
Sounds good, though I would prefer `output_variables` instead of `solution_include` to match the syntax of `QuickPlot`

martinjrobins, 2023-08-23:
Note that we probably need to do this for the JaxSolver as well, see https://github.com/pybop-team/PyBOP/pull/25

BradyPlanden, 2023-08-24:
Just to crosspost, I'll start working on the Jax implementation of this as discussed in the above PyBOP issue.

PR #3217 (2023-09-29): Idaklu solver can be given a list of variables to calculate during the solve
# Description

Provides a mode for the (Idaklu) solver so that only a user given list of variables are calculated and stored as part of the solution. This functionality extends to sensitivity calculations. The PR also includes a restructuring of the idaklu c++ codebase to support further extensions and abstractions in the future.

Fixes #2796 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include  [... cut]