pybamm-team/PyBaMM #2811: Hello，Can you provide a parameter set for "Marquis2020" and "Yang2017"

IDontIikeHandwriting, 2023-03-26:
### Discussed in https://github.com/pybamm-team/PyBaMM/discussions/2798

<div type='discussions-op-text'>

<sup>Originally posted by **IDontIikeHandwriting** March 23, 2023</sup>
The sei_growth.py（https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/interface/sei/sei_growth.py） uses the "Yang2017" and "Marquis2020" parameter sets, but the input section of the GitHub file （https://github.com/pybamm-team/PyBaMM/tree/develop/pybamm/input/parameters/lithium_ion） does not provide these two parameter sets. Can you upload these two parameter sets to the GitHub? Thank you very much.</div>

arjxn-py, 2023-03-26:
Hi, @IDontIikeHandwriting I cannot acess the files that you have referenced with the URLs. Can you please look into it?
Thank You.

tinosulzer, 2023-03-26:
It adds references to those papers (you can find the full details in `CITATIONS.txt` or by running `pybamm.print_citations()` at the end of a python file where you used the SEI growth model). That is different from parameter sets. There are no parameter sets for Marquis2020 and Yang2017 (we tried adding a parameter set for Yang2017, but they purposefully keep some of the parameters secret). Most of the parameter sets have some SEI parameters

closed 2023-03-26 by tinosulzer

IDontIikeHandwriting, 2023-03-26:
> It adds references to those papers (you can find the full details in `CITATIONS.txt` or by running `pybamm.print_citations()` at the end of a python file where you used the SEI growth model). That [...]

Thanks for your reply！

IDontIikeHandwriting, 2023-03-26:
> Hi, @IDontIikeHandwriting I cannot acess the files that you have referenced with the URLs. Can you please look into it once? Thank You.

Hello, you can see these files in the code section of GitHub