pybamm-team/PyBaMM #2813: adding `nbQA` support for example notebooks

agriyakhetarpal, 2023-03-26:
### Description

[nbQA](https://nbqa.readthedocs.io/en/latest/examples.html#command-line-examples) now includes support for `ruff` to format Jupyter notebooks and already has integration for `black`, so we can add it for both (or even just`ruff` would be fine)

### Motivation

We already use `ruff` as a pre-commit hook

### Possible Implementation

in `.pre-commit-config.yaml`

### Additional context

From "Does Ruff support Jupyter notebooks? -- https://beta.ruff.rs/docs/faq/#does-ruff-support-jupyter-notebooks"

arjxn-py, 2023-03-26:
I went through the resources you have attached & I want to give it a try.
Furthermore I feel that I might have doubts while working on it which I'll be clearing here itself.

Saransh-cpp, 2023-03-27:
@arjxn-py this is more of a discussion at this moment. We want to see how the notebooks will evolve during GSoC and then decide if we want these pre-commits or not.

arjxn-py, 2023-03-27:
Got it @Saransh-cpp.

Saransh-cpp, 2023-06-09:
@agriyakhetarpal I think this aligns with your GSoC project? Feel free to pick this up. @arjxn-py can also pick this up as it relates to the infrastructure project as well.

arjxn-py, 2023-06-09:
Happy to take this up.

agriyakhetarpal, 2023-06-10:
That's nice. I discussed embedding the example notebooks on readthedocs during today's meeting (which is why I was waiting on this), so I will be restructuring some parts of the documentation in a PR soon — @arjxn-py I would recommend starting with this after that gets merged

Saransh-cpp, 2023-06-27:
@arjxn-py it would be nice to add some pre-commits for notebooks once #3043 is merged. You can maybe pick this up once the option dependencies PR is merged.

PR #3110 (2023-07-22): Add `nbqa` support support for example notebooks
# Description

Adding nbqa as pre-commit hook for jupyter notebooks.

Fixes #2813 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checkli [... cut]
closed 2023-07-22 by Saransh-cpp