pybamm-team/PyBaMM #2827: Issues about  parameter "Number of cells connected in series to make a battery" with model equivalent_circuit.Thevenin

whblcy, 2023-03-29:
### Discussed in https://github.com/pybamm-team/PyBaMM/discussions/2826

<div type='discussions-op-text'>

<sup>Originally posted by **whblcy** March 29, 2023</sup>
Hi， when I use the equivalent circuit model, setting the parameters of the cell numbers always does not take effect. Why is this? How do I modify it.
code as below:

`import pybamm
import numpy as np

model = pybamm.equivalent_circuit.Thevenin()
experiment = pybamm.Experiment(
    [
        "Discharge at 10W for 1.2 hours or until 3.5 V",
    ]
)

parameter_values = model.default_parameter_values

parameter_values.update({"Number of electrodes connected in parallel to make a cell": 2}, check_already_exists=False)
parameter_values.update({"Number of cells connected in series to make a battery": 16}, check_already_exists=False)

sim = pybamm.Simulation(model, experiment=experiment, parameter_values=parameter_values)
sim.solve()
options = [
        "Current [A]",
        "Battery voltage [V]",
        "Terminal voltage [V]",
        "Open circuit voltage [V]",
    ]
sim.plot(options)`

the result as below:
[image]

</div>

Scottmar93, 2023-03-29:
At the moment, that parameter has not been implemented with the ECM. Currently available parameters for adjustment in the ECM are can be seen in the example set: 

https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/input/parameters/ecm/example_set.py

It shouldn’t be super hard to add it though as I think you could just scale the currents, capacities, etc in the appropriate manner within here https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/parameters/ecm_parameters.py

If you want to pick that up, go for it :) 

You could also just scale your currents and capacities before putting them into the simulation as well if you want a workaround.

whblcy, 2023-03-29:
Thank you  Scottmar. I will try with your idea.

tinosulzer, 2023-05-08:
@whblcy any update on this?

closed 2023-06-10 by tinosulzer