pybamm-team/PyBaMM #2839: [Bug]: Sign error in Ohmic loss variable?

DavidMStraub, 2023-03-30:
### PyBaMM Version

22.12

### Python Version

3.8.10

### Describe the bug

I noticed that `X-averaged battery solid phase ohmic losses [V]` is always smaller in magnitude than the individual electrode Ohmic losses `X-averaged {domain} electrode ohmic losses [V]`, but I expected it to be the sum. I believe this is a sign error in the definition of `delta_phi_s_av` in `base_battery_model`, unless I am misunderstanding the purpose of this variable.

Here is an illustrative plot (sorry for not providing a more complete code example):

[image]

### Steps to Reproduce

```python
plt.plot(sol["Time [s]"].entries, 1e3 * sol["X-averaged positive electrode ohmic losses [V]"].entries / sol["Current [A]"].entries, label="PE")
plt.plot(sol["Time [s]"].entries, 1e3 * sol["X-averaged negative electrode ohmic losses [V]"].entries / sol["Current [A]"].entries, label="NE")
plt.plot(sol["Time [s]"].entries, 1e3 * sol["X-averaged battery solid phase ohmic losses [V]"].entries / sol["Current [A]"].entries, label="tot")
plt.legend()
plt.grid()
```

tinosulzer, 2023-04-03:
Can you check with the latest version whether this still persists? I made some changes to this recently. Also check https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/plotting/plot_voltage_components.ipynb for the intended signs

tinosulzer, 2023-05-08:
Closing due to 30 days with no reply. Feel free to reopen if needed

closed 2023-05-08 by tinosulzer