pybamm-team/PyBaMM #2858: Updates for casadi 3.6

tinosulzer, 2023-04-05:
Release notes: https://github.com/casadi/casadi/releases/tag/3.6.0

Will add as found

- time grid points should be passed when creating interpolator instead of when solving
- rootfinder errors

tinosulzer, 2023-04-05:
I don't know how to fix the failing sensitivity test so skipping for now, @martinjrobins @jsbrittain could you take a look at this?

martinjrobins, 2023-04-18:
sure, this is also reported in https://github.com/pybamm-team/PyBaMM/issues/2789#issuecomment-1482739316.

tinosulzer, 2023-04-19:
I think it might be a different error here

martinjrobins, 2023-04-26:
Did PR #2883 fix the problem? I just tried `test_sensitivities` on spm, spme and dfn with idaklu solver and it seemed to work ok.

martinjrobins, 2023-04-26:
ahh ok, I wasn't aware you'd added that

martinjrobins, 2023-04-27:
near as I can figure this is the problem. Say you have a model like this

$$\frac{du}{dt} = - a * u$$

$$ v = u $$

$$ u(0) = 1$$

$$ v(0) = 1$$

For $a=1$, sundials calculates the initial condition for $s = \frac{dv}{da}$ as $s(0) = [0, 0]$, presumably because v(0) does not depend on $a$, but if you correctly write the initial condition for $v$ as:

$$\frac{du}{dt} = - a * u$$

$$ v = u $$

$$ u(0) = 1$$

$$ v(0) = a$$

Then sundials finds the initial condition should be $s(0) = [0, 1]$. Not exactly sure what the expected behaviour is really, or why sundials isn't calculating the correct ic for $s$ even when you don't give the correct symbolic form for $v(0)$

tinosulzer, 2023-04-27:
$s(0) = [0, 1]$ is the correct initial condition in the second case, right?

martinjrobins, 2023-04-29:
> s(0)=[0,1] is the correct initial condition in the second case, right?

yup, and I believe it should be for the 1st as well, so that is why I'm a bit confused that the `calcic` function in sundials (ie. calculate consistent initial conditions) isn't finding it, it just returns `[0, 0]` for the 1st and `[0, 1]` for the 2nd.

martinjrobins, 2023-05-03:
ahh ok. You have to call `IDAGetSens` after `IDACalcIC` in order for the solution to actually show up in your sensitivity vector..... fixed now

PR #2920 (2023-05-19): I2858-init-cond-sundials
# Description

1. feed initial conditions for the sensitivity vectors to the idaklu solver (before was just initialised to 0)
2. call idagetsens after idacalcic to get the corrected initial conditions for sensitivity vectors
3. make the FD h a bit smaller to reduce numerical errors in sensitivity validation for models

Fixes #2858

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-and [... cut]