pybamm-team/PyBaMM #2866: Using Pybamm to identify ECM parameters

SupremeZ01, 2023-04-07:
### Description

Can I treat Pybamm's DFN model as a real battery and use GITT or HPPC testing to identify the SOC-OCV curve and electrical parameters of the first-order equivalent circuit model?
I tried to do this work, but found that the error between the identified ECM model and Pybamm was a bit large. Especially when I used the ECM model and Pybamm for online closed-loop control, I found that the error between the SOC of my ECM model and the SOC calculated by Pybamm was a bit large. I don't know how to solve this problem.

SupremeZ01, 2023-04-12:
### Thanks for your attention!
 I used the eSOH model to calculate the SOC. The version of Pybamm is 22.9.
```
import pybamm

model = pybamm.lithium_ion.DFN()
parameter_values = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Chen2020)
experiment = pybamm.Experiment([
    ("Rest for 1 hour (10 s period)",
    "Charge at 0.05C until 4.2V (10 s period)",
    "Rest for 1 hour (10 s period)",
    "Discharge at 0.05C until 2.5V (10 s period)",)
])
sim = pybamm.Simulation(model, experiment=experiment, parameter_values=parameter_values)
sol = sim.solve(initial_soc = 0)

param = pybamm.LithiumIonParameters()
V_min = parameter_values.evaluate(param.voltage_low_cut_dimensional)
V_max = parameter_values.evaluate(param.voltage_high_cut_dimensional)
C_n = parameter_values.evaluate(param.n.cap_init)
C_p = parameter_values.evaluate(param.p.cap_init)
n_Li = parameter_values.evaluate(param.n_Li_particles_init)

esoh_solver = pybamm.lithium_ion.ElectrodeSOHSolver(parameter_values, param)

inputs = {"V_min": V_min, "V_max": V_max, "C_n": C_n, "C_p": C_p, "n_Li": n_Li}

esoh_sol = esoh_solver.solve(inputs)
x_0 = esoh_sol["x_0"].data[0]
x_100 = esoh_sol['x_100'].data[0]
y_0 = esoh_sol["y_0"].data[0]
y_100 = esoh_sol['y_100'].data[0]
c_n_max = parameter_values.evaluate(param.n.prim.c_max)
c_p_max = parameter_values.evaluate(param.p.prim.c_max)

x = sol['Average negative particle concentration'].entries
SOC = (x-x_0)/(x_100-x_0)
```
### Supplement:
I used  GITT testing to get OCV-SOC curve:
```
model = pybamm.lithium_ion.DFN()  
parameter_values = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Chen2020)

experiments = pybamm.Experiment(
    [("Rest for 1 hour (10 s period)",
    "Charge at 0.05C until 4.2V (10 s period)",
    "Rest for 1 hour (10 s period)",
    "Discharge at 0.05C until 2.5V (10 s period)",
    )])

initial_soc = 0
sim = pybamm.Simulation(model, experiment=experiments, parameter_values=parameter_values)
sol = sim.solve(initial_soc=initial_soc)
```
Also,  I used HPPC testing to obtain the R0, R1, C1 parameter values ​​of the ECM model:
```
experiment = pybamm.Experiment(["Rest for 1 hour"] + [
    ("Rest for 10 s (1 s period)",
    f"Discharge at 1C for 10 s (0.1 s period)",
    "Rest for 40 s (0.2 s period)",
    f"Charge at 0.75C for 10 s (0.1 s period)",
    f"Rest for 40 s (0.2 s period)"),
],
# period = "0.2 s", 
termination = "2.5 V"
)

initial_socs = [1, 0.95, 0.9, 0.8, 0.7, 0.6, 0.5, 0.4, 0.3, 0.2, 0.1]
model = pybamm.lithium_ion.DFN()       
parameter_values = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Chen2020)
parameter_values["Ambient temperature [K]"] = 298.15

sols = []
sols_plot = []
for initial_soc in initial_socs:
    sim = pybamm.Simulation(model, experiment=experiment, parameter_values=parameter_values)
    sol = sim.solve(initial_soc=initial_soc)
    sol = sol.cycles[1]
    sols_plot.append(sol)
    sols.append({sol:str(initial_soc)})

for i,sol in enumerate(sols_plot):
        t = sol["Time [s]"].data
        t = t - t[0]
        V = sol["Terminal voltage [V]"].data
        plt.plot(t, V)
```
The output of HPPC testing:
[image]

So, I guess that the R0, R1, and C1 parameter values I identified with HPPC are wrong, so my ECM model cannot track Pybamm's DFN model. Maybe you can give me some other suggestion for model identification, thanks for your help.

tinosulzer, 2023-04-13:
Hopefully someone more familiar with ECMs can help. I'll convert this to a discussion since I don't think there are any associated bugs or feature requests