pybamm-team/PyBaMM #2878: [Bug]: Parameter Positive/Negative electrode charge transfer coefficient inconsistent with the true value

1836005678, 2023-04-15:
### PyBaMM Version

23.3

### Python Version

3.9.13

### Describe the bug

Parameter Positive/Negative electrode charge transfer coefficient inconsistent with the true value
These two parameters **can be edited** by (take parameter set Prada2013 as an example):
`param = pybamm.ParameterValues("Prada2013")`
`param["Negative electrode charge transfer coefficient"] = 0.7 * param["Negative electrode charge transfer coefficient"]`
`param["Positive electrode charge transfer coefficient"] = 0.3 * param["Positive electrode charge transfer coefficient"]`
However, the true value of these two parameters during calculation still **remain to be the original value (0.5)** **due to the code implementation** in function LFP_electrolyte_exchange_current_density_kashkooli2017 (positive electrode electrolyte exchange_current_density) and graphite_LGM50_electrolyte_exchange_current_density_Chen2020 (negative electrode electrolyte exchange_current_density) in Prada2013.py:
`def LFP_electrolyte_exchange_current_density_kashkooli2017(c_e, c_s_surf, c_s_max, T):`
`    return (
        m_ref * arrhenius * c_e**0.5 * c_s_surf**0.5 * (c_s_max - c_s_surf) ** 0.5
    )`
[image]
[image]
`def graphite_LGM50_electrolyte_exchange_current_density_Chen2020(
    c_e, c_s_surf, c_s_max, T
):`
`    return (
        m_ref * arrhenius * c_e**0.5 * c_s_surf**0.5 * (c_s_max - c_s_surf) ** 0.5
    )`
[image]
[image]
Calculation result also show that the true value would not be changed through the above method as all result are completely same.
[image]
[image]
**This bug may cause user misuse.
This bug also exists in other parameter sets.**

For parameter set ORegan2022, even the nominal value is incorrect.
(0.5-0.43)
[image]
[image]
(0.5-0.792)
[image]
[image]

Although I can edit true values in electrodes electrolyte exchange current density function through my own code implementation, i can not update the nominal value at the same time. Let talk about how to update the true value along with the nominal value at the same time.

### Steps to Reproduce

Jupyter Notebook code for verification the above phenomenon:
%pip install pybamm -q    # install PyBaMM if it is not installed
import pybamm

x = "Negative electrode charge transfer coefficient" # or "Positive electrode charge transfer coefficient"
model = pybamm.lithium_ion.DFN()
param1 = pybamm.ParameterValues("Prada2013")
param2 = pybamm.ParameterValues("Prada2013")
param2[x] = 0.7 * param2[x]
param3 = pybamm.ParameterValues("Prada2013")
param3[x] = 0.3 * param3[x]
params = [param1,param2,param3]

sims = []
for param in params:
    sim = pybamm.Simulation(model = model, parameter_values = param)
    sim.solve([0, 3600])
    sims.append(sim)

pybamm.dynamic_plot(sims, time_unit="seconds")

print(sims[0].solution["Voltage [V]"].entries)
print(sims[1].solution["Voltage [V]"].entries)
print(sims[2].solution["Voltage [V]"].entries)

tinosulzer, 2023-04-26:
I don't follow. Users can set their own transfer coefficient and their own exchange-current density function. As with many other things in PyBaMM, it's possible to do it wrong if you don't know what you're doing. What would you suggest we do instead?

1836005678, 2023-05-04:
I agree with you. I think it is reasonable to delete these two paramerter from parameter sets to avoid misunderstanding, as long as this action would not make any mistake.

1836005678, 2023-05-06:
"Negative electrode charge transfer coefficient"
"Positive electrode charge transfer coefficient"

Search results for keywords in the pybamm folder:
[image]
They are only assigned in parameter sets, but not used even once in pybamm simulation.

tinosulzer, 2023-05-08:
They are used if certain model options are selected (asymmetric kinetics). This is the same for some SEI and thermal parameters (not used by default options but used if SEI or thermal options are activated). We don't currently have plans to change that, so I'm going to close this issue. See also #2341

closed 2023-05-08 by tinosulzer

1367077806, 2023-05-18:
Hello! Where can I search these parameters like you by using pybamm folder? How can I find this function?

1836005678, 2023-05-23:
> Hello! Where can I search these parameters like you by using pybamm folder? How can I find this function?

Functions and parameters can be fould in "\pybamm\input\parameters\lithium_ion\xx_20xx.py".

1367077806, 2023-05-23:
同学你也是在进行pybamm建模的吗，如果可以的话，方便加个微信交流一下？我是厦门大学研一的学生。

刘炯严
***@***.***

                       
原始邮件

发件时间：2023/5/23 15:56

收件人："pybamm-team/PyBaMM"< ***@***.*** &gt;;

抄送人："1367077806"< ***@***.*** &gt;;"Comment"< ***@***.*** &gt;;