pybamm-team/PyBaMM #2884: Apple M-Series Self-Hosted Runner

BradyPlanden, 2023-04-19:
### Description

@martinjrobins and I have been able to purchase an Apple M2 Mini to utilise as a self-hosted GitHub runner for the PyBaMM. I'm currently completing the initial setup and once we have a robust connection to the Oxford Uni network I can start on the runner environment setup. 

For now, I'll do my best to ensure it is updated and online. Open to other thoughts and interests, however!

### Motivation

Supports #2346

### Possible Implementation

I suspect I can follow this [guide](https://betterprogramming.pub/run-github-actions-self-hosted-macos-runners-on-apple-m1-mac-b559acd6d783) for most of the setup. I'll need access to the link the runner to PyBaMM repository.

martinjrobins, 2023-04-21:
Thanks @BradyPlanden, I've added you to the repo with write access (you should have an invite), is that all you'll need?

BradyPlanden, 2023-04-21:
The runner is now live as `Mac-Mini-M2-Pro`. The next step is to integrate it into the current workflows. The nightly test workflow `test_on_push.yml ` will be completed first.

BradyPlanden, 2023-05-15:
Discussion from developers meeting:

- Implement the runner for cron jobs only
- Harden the workflows' folder via `CODEOWNERS` group
- Integrate the runner onto an isolated network

PR #2929 (2023-07-10): Add self-hosted actions to periodic tests
# Description

This updates the `run_periodic_tests.yml` workflow to include the self-hosted m-series runner.

Fixes #2884 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change wh [... cut]