pybamm-team/PyBaMM #2897: [Bug]: Install from source for M-Series hardware

BradyPlanden, 2023-04-26:
### PyBaMM Version

Source

### Python Version

3.10 & 3.11

### Describe the bug

Installing from source results in the following sundials header error via the `brew install sundials` method for M-series hardware. Using, `python3 -m tox -e pybamm-requires` fixes the issue. It appears to be a linking issue, as the `brew install sundials` doesn't appear to install libraries in `~/.local`. Seems close to #1296

### Steps to Reproduce

Follow MacOS installation via https://pybamm.readthedocs.io/en/stable/source/user_guide/installation/install-from-source.html#install-from-source-developer-install

### Relevant log output

```shell
PyBaMM git:(self-hosted-workflow) python3 -m tox -e dev
dev create: /Users/bradyplanden/Documents/Git/PyBaMM/.tox/dev
dev installdeps: cmake, black, scikits.odes
ERROR: invocation failed (exit code 1), logfile: /Users/bradyplanden/Documents/Git/PyBaMM/.tox/dev/log/dev-1.log
========================================================= log start ==========================================================
Collecting cmake
  Using cached cmake-3.26.3-py2.py3-none-macosx_10_10_universal2.macosx_10_10_x86_64.macosx_11_0_arm64.macosx_11_0_universal2.whl (45.6 MB)
Collecting black
  Using cached black-23.3.0-cp311-cp311-macosx_10_16_universal2.whl (2.6 MB)
Collecting scikits.odes
  Using cached scikits.odes-2.7.0.tar.gz (931 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting click>=8.0.0 (from black)
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting mypy-extensions>=0.4.3 (from black)
  Using cached mypy_extensions-1.0.0-py3-none-any.whl (4.7 kB)
Collecting packaging>=22.0 (from black)
  Using cached packaging-23.1-py3-none-any.whl (48 kB)
Collecting pathspec>=0.9.0 (from black)
  Using cached pathspec-0.11.1-py3-none-any.whl (29 kB)
Collecting platformdirs>=2 (from black)
  Using cached platformdirs-3.3.0-py3-none-any.whl (15 kB)
Collecting scipy (from scikits.odes)
  Using cached scipy-1.10.1-cp311-cp311-macosx_12_0_arm64.whl (28.7 MB)
Collecting numpy<1.27.0,>=1.19.5 (from scipy->scikits.odes)
  Using cached numpy-1.24.3-cp311-cp311-macosx_11_0_arm64.whl (13.8 MB)
Building wheels for collected packages: scikits.odes
  Building wheel for scikits.odes (pyproject.toml): started
  Building wheel for scikits.odes (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error

  × Building wheel for scikits.odes (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [2345 lines of output]
      <string>:25: DeprecationWarning:

        `numpy.distutils` is deprecated since NumPy 1.23.0, as a result
        of the deprecation of `distutils` itself. It will be removed for
        Python >= 3.12. For older Python versions it will remain present.
        It is recommended to use `setuptools < 60.0` for those Python versions.
        For more details, see:
          https://numpy.org/devdocs/reference/distutils_status_migration.html

      /private/var/folders/46/vvrt_nq91kj6lc8nv1_p7p6r0000gn/T/pip-build-env-zee0hxd8/overlay/lib/python3.11/site-packages/setuptools/dist.py:286: SetuptoolsDeprecationWarning: The namespace_packages parameter is deprecated, consider using implicit namespaces instead (PEP 420).
        warnings.warn(msg, SetuptoolsDeprecationWarning)
      /private/var/folders/46/vvrt_nq91kj6lc8nv1_p7p6r0000gn/T/pip-build-env-zee0hxd8/overlay/lib/python3.11/site-packages/setuptools/dist.py:771: UserWarning: Usage of dash-separated 'description-file' will not be supported in future versions. Please use the underscore name 'description_file' instead
[... 531 lines cut ...]
        Compiler    : clang

      CPU baseline  :
        Requested   : 'min'
        Enabled     : NEON NEON_FP16 NEON_VFPV4 ASIMD
        Flags       : none
        Extra checks: none

      CPU dispatch  :
        Requested   : 'max -xop -fma4'
        Enabled     : ASIMDHP ASIMDDP ASIMDFHM
        Generated   : none
      INFO: CCompilerOpt.cache_flush[857] : write cache to path -> /private/var/folders/46/vvrt_nq91kj6lc8nv1_p7p6r0000gn/T/pip-install-o25_jc9c/scikits-odes_e5f5f73cf9184453b0bc799385f15735/build/temp.macosx-13-arm64-cpython-311/ccompiler_opt_cache_ext.py
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for scikits.odes
Failed to build scikits.odes
ERROR: Could not build wheels for scikits.odes, which is required to install pyproject.toml-based projects

========================================================== log end ===========================================================
ERROR: could not install deps [cmake, black, scikits.odes]; v = InvocationError('/Users/bradyplanden/Documents/Git/PyBaMM/.tox/dev/bin/python -m pip install cmake black scikits.odes', 1)
__________________________________________________________ summary ___________________________________________________________
ERROR:   dev: could not install deps [cmake, black, scikits.odes]; v = InvocationError('/Users/bradyplanden/Documents/Git/PyBaMM/.tox/dev/bin/python -m pip install cmake black scikits.odes', 1)
```

jsbrittain, 2023-04-28:
To confirm, this is reproducible on both mac and linux when installing via the brew method. At the moment a quick fix is to specify the following environment variables prior to running `tox -e dev`:
```
export CMAKE_PREFIX_PATH="$(brew --prefix)"
export SUNDIALS_INST="$(brew --prefix sundials)"
```
I'm just looking into a cleaner solution...

jsbrittain, 2023-05-04:
Yes, I think directing them to `pybamm-requires` seems the most straightforward and provides a clear recommended developer procedure, and has already been noted to solve the original Issue as raised by @BradyPlanden .

Just note that the Mac install does still rely on `brew` for `openblas`, etc., and when testing this approach (admittedly in a linux container, i.e. not the recommended approach), `pybamm-requires` then had problems finding `blas` until directed towards the brew install. Testing on Mac with `brew` installed to the default location seemed to work fine though.

I'll update and simplify the install docs in accordance with @agriyakhetarpal suggestion above, specifically point 2.

PR #2925 (2023-05-08): Remove brew-sundials from the recommended developer installations
# Description

The (Mac recommended) `brew` install procedure for sundials does not link the cmake args that are required by `tox -e dev`. Here, we simplify the developer installation procedure for Mac, bringing it in-line with that for linux, by using `tox -e pybamm-requires` to download, compile and install the SuiteSparse and SUNDIALS libraries.

Fixes #2897 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change ( [... cut]
closed 2023-05-08 by tinosulzer