pybamm-team/PyBaMM #2901: Deprecate old ways of creating ParameterValues

tinosulzer, 2023-04-26:
I think the transition to using python files or json file (BPX) for parameter values has gone fairly well so we can start to deprecate some of the old methods for importing values from csv files. These are no longer recommended but still possible from the codebase. We should also update the training videos on this very soon.
- [ ] remove functionality to update from chemistry (old approach with separate files for anode/cathode/electrolyte etc)
- [ ] remove functionality to load from csv
- [ ] remove functionality to load a function from a string "[function]function_name"
- [ ] remove functionality to load data from a string "[data]data_name"

Basically, users will be responsible for passing in a python dictionary where functions are python functions. Interpolants can be created as functions [like this](https://github.com/pybamm-team/PyBaMM/blob/18d4e84d07a8afb1d8a71081c2912f486b11283d/pybamm/input/parameters/lithium_ion/Ai2020.py#L501-L517)

This is a breaking change so let's discuss before committing

brosaplanella, 2023-04-27:
Shall we keep these options with a warning for a few releases and then remove?

tinosulzer, 2023-04-27:
We've had this warning up for ~6 releases https://github.com/pybamm-team/PyBaMM/blob/18d4e84d07a8afb1d8a71081c2912f486b11283d/pybamm/parameters/parameter_values.py#L65

brosaplanella, 2023-04-27:
Then happy to deprecate it :)

PR #2959 (2023-05-22): Issue 2901 parameters csv
# Description

In recent versions of PyBaMM, we recommend using python files to specify parameter sets (as opposed to csv files in early versions). This PR removes all the functionality that was included to make loading from csv files possible. Loading from csv files is no longer possible, and Python files should now always be used instead.

Fixes #2901 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If  [... cut]
closed 2023-05-22 by tinosulzer