pybamm-team/PyBaMM #2908: `inspect` causing big performance regression

tinosulzer, 2023-04-30:
[image]
[image]

This is due to #2862 

@agriyakhetarpal can you look into a way to make this faster? Otherwise we should revert, it's not worth the functionality gain

tinosulzer, 2023-04-30:
This is a huge performance hit in some cases (e.g. repeated solving of the SPM is not 10x slower) so we should fix immediately and make a 23.4.1 release
[image]

Maybe this helps https://stackoverflow.com/questions/17407119/python-inspect-stack-is-slow

tinosulzer, 2023-04-30:
Also weird that the benchmarks never caught this, are they working now @Saransh-cpp ?

agriyakhetarpal, 2023-04-30:
Since this is immediate I am happy to revert — though I will get back on this and try some other faster implementation in the next week, wasn't aware of such a potential slowdown 😅 Where can I run the above profiler?

tinosulzer, 2023-04-30:
It's snakeviz, you can pip install then run in `ipython` terminal 
```
> %load_ext snakeviz
> %snakeviz function_to_profile()
```

tinosulzer, 2023-04-30:
Potential solutions are:
- find a faster implementation of doing the same thing
- use some form of caching so this is only calculated once per `register` call
- add tagging manually as an input to the `register` function
- remove altogether

agriyakhetarpal, 2023-05-01:
I agree with removing altogether and releasing a patch for now.

- We can indeed explore a caching option, albeit will some overhead still remain when a model is called and its citation is registered for the first time when running an experiment? In that case a faster `inspect.stack()` implementation altogether would be better or by modifying all `__call__` methods with a metaclass

- Preserving a `dict` of models mapped to their citations hardcoded manually is also a solution 

I am just figuring out some end-semester assignments and submissions so it will take me a few days to try these ways. Please go ahead and reopen #1182 for the time being

closed 2023-05-02 by tinosulzer

Saransh-cpp, 2023-05-02:
> Also weird that the benchmarks never caught this, are they working now @Saransh-cpp ?

The benchmarks were indeed failing on develop, but the benchmark workflow was fixed after the `inspect` PR was merged. Also, benchmarks only run on `push` and would not run on PRs from forks (I am not sure why?).

tinosulzer, 2023-05-02:
We should allow it to run on PRs from forks

agriyakhetarpal, 2023-05-05:
@tinosulzer, I tried to profile a faster implementation (it turns out that using `inspect.currentframe` is miles faster than `inspect.stack` because it does not create a stack trace). However, I am not sure if I'm correctly interpreting the visualisations I get from `snakeviz` – here's what I am running:

```python
# In the ipython shell
import pybamm
%load_ext snakeviz
%snakeviz pybamm.lithium_ion.SPMe()
```
## Before
[image]

and the results for `pybamm.citations.register()`, and for `_caller_name()` that gets called internally by `register()`
| ncalls | tottime | percall | cumtime  | percall  | filename:lineno(function)     |
|--------|---------|---------|----------|----------|-------------------------------|
| 1 | 7.4e-06 | 7.4e-06 | 0.03022 | 0.03022 | citations.py:109(register)    |
| 1 | 9.9e-06 | 9.9e-06 | 0.03021  | 0.03021  | citations.py:61(_caller_name) |

## Now

[image]

| ncalls | tottime | percall | cumtime  | percall  | filename:lineno(function)     |
|--------|---------|---------|----------|----------|-------------------------------|
| 1 | 6.4e-06 | 6.4e-06 | 1.35e-05 | 1.35e-05 | citations.py:111(register)    |
| 1 | 4.2e-06 | 4.2e-06 | 6.4e-06   | 6.4e-06    | citations.py:61(_caller_name) |

which seems to be at least 2000x faster if we compare cumulative times (the earlier takes ~27% of the time taken to call SPM, which then dropped down to 0.02% in the latter); assuming I am reading the results right. This can be sped up even more if we use `sys._getframe` instead, but using internal functions from `sys` is generally discouraged. 

Could you perhaps write the code for a longer and very complex PyBaMM experiment that I can try to profile (such as the ones running behind your screenshots above) just to see if the results are scaling properly and do not cause any performance regressions?

tinosulzer, 2023-05-05:
```python
> import pybamm
> %load_ext snakeviz
> %snakeviz model = pybamm.lithium_ion.DFN()
> sim = pybamm.Simulation(model) # should be very fast, no need to profile
> %snakeviz sim.solve([0,3600]) 
> %snaleviz sim.solve([0,3600])
```
The final solve is repeated twice because several things get cached the first time. Ideally, the second time, almost all the time should be spent in the `_integrate` method of the solver.

Also repeat the whole thing with `pybamm.lithium_ion.SPM()`, which is faster and therefore any overheads are a larger percentage of the total runtime.

It looks like using `inspect.currentframe` will be a sufficient fix.

agriyakhetarpal, 2023-05-06:
Thanks; for initialising the DFN model, the performance hit with `register` is 0.01–0.03%. With `sim.solve([0, 3600])`, the mean comes out to be around 0.02%, and on running `sim.solve([0, 3600])` again, it comes out as about 0.15% (15 ncalls). I repeated the process for ten iterations (for SPM as well below) to make sure I'm not receiving any statistical outliers and that the behaviour was okay. 

The second time, `_integrate` always takes ~94–95% of the total time, which, as you said, should be ideal:

[image]

Similar results arise for the SPM model: the performance hit for `register` is 0.01–0.03%, though on the second iteration of `sim.solve([0, 3600])`, it shoots up to 1.5–1.6% (14 ncalls). Since things are cached, `_integrate` uses 88% of the total time:

[image]

which means registering citations for the SPM model impacts it just slightly more than doing so for the DFN model – is this performance reasonable enough to open a PR, or should I try to improve upon this more?

tinosulzer, 2023-05-16:
Sorry I forgot to reply to previous discussion. The speedup is great, but it's surprising that it's being called 14 times on the second solve. Can you find a way to cache the citation tags?

agriyakhetarpal, 2023-05-16:
Yes sure, I'll give that a try and convert that to a draft – would you suggest using `functools` or would some basic caching with a dictionary suffice? If we're aiming for a best-case scenario, the time taken to read the citation from it should hopefully be negligible compared to registering it

tinosulzer, 2023-05-16:
Either is fine

agriyakhetarpal, 2023-05-16:
Just using `@cache` from `functools` (under the assumption that a built-in way would be better than writing a custom cache) and without any other changes halves the total time taken to about ~0.87% (averaged out across 10 times) on the second time a simulation of the SPM model is solved with `sim.solve([0, 3600])`, and this value keeps getting smaller in later iterations, but the number of ncalls remains the same due to some reason. 

[image]

PR #2909 (2023-05-02): #2908 revert citations tag
# Description

Fixes #2908 
Prepares release 23.4.1

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ pre-commit run` (see [ [... cut]