pybamm-team/PyBaMM #2914: BPX diffusivity interpolant issue[Bug]:

chenzhao-py, 2023-05-02:
### PyBaMM Version

22

### Python Version

3.9

### Describe the bug

I have used BPX json file for the parameter sets. For the positive diffusivity, I have the table below:

 "Diffusivity [m2.s-1]": {
                        "x" : [0.01400862, 0.02383751, 0.03366641, 0.0434953 , 0.05332419,
                        0.06315308, 0.07298198, 0.08281087, 0.09263976, 0.10246865,
                        0.11229754, 0.12212644, 0.13195533, 0.14178422, 0.15161311,
                        0.16144201, 0.1712709 , 0.18109979, 0.19092868, 0.20075758,
                        0.21058647, 0.22041536, 0.23024425, 0.24007315, 0.24990204,
                        0.25973093, 0.26955982, 0.27938872, 0.28921761, 0.2990465 ,
                        0.30887539, 0.31870428, 0.32853318, 0.33836207, 0.34819096,
                        0.35801985, 0.36784875, 0.37767764, 0.38750653, 0.39733542,
                        0.40716432, 0.41699321, 0.4268221 , 0.43665099, 0.44647989,
                        0.45630878, 0.46613767, 0.47596656, 0.48579545, 0.49562435,
                        0.50545324, 0.51528213, 0.52511102, 0.53493992, 0.54476881,
                        0.5545977 , 0.56442659, 0.57425549, 0.58408438, 0.59391327,
                        0.60374216, 0.61357106, 0.62339995, 0.63322884, 0.64305773,
                        0.65288663, 0.66271552, 0.67254441, 0.6823733 , 0.69220219,
                        0.70203109, 0.71185998, 0.72168887, 0.73151776, 0.74134666,
                        0.75117555, 0.76100444, 0.77083333, 0.78066223, 0.79049112,
                        0.80032001, 0.8101489 , 0.8199778 , 0.82980669, 0.83963558,
                        0.84946447, 0.85929337, 0.86912226, 0.87895115, 0.88878004,
                        0.89860893, 0.90843783, 0.91826672, 0.92809561, 0.9379245 ,
                        0.9477534 , 0.95758229, 0.96741118, 0.97724007, 0.98706897],
                        "y" : [2.71105000e-16, 2.69469859e-16, 2.76328269e-16, 2.92348699e-16,
                        3.18199620e-16, 3.54549500e-16, 4.02066812e-16, 4.61420023e-16,
                        5.33277605e-16, 6.19652775e-16, 7.30126250e-16, 8.76923723e-16,
                        1.07227366e-15, 1.32813401e-15, 1.64543302e-15, 2.01046701e-15,
                        2.40852323e-15, 2.82488894e-15, 3.24832709e-15, 3.69435552e-15,
                        4.19091456e-15, 4.76293695e-15, 5.39075200e-15, 6.02479570e-15,
                        6.69118967e-15, 7.42363658e-15, 7.89132178e-15, 8.30192407e-15,
                        1.04204198e-14, 1.18014726e-14, 1.37795651e-14, 1.57736204e-14,
                        1.75247795e-14, 2.01973001e-14, 2.28902286e-14, 2.54146463e-14,
                        2.79039107e-14, 3.01729853e-14, 3.22174016e-14, 3.39368345e-14,
                        3.52211003e-14, 3.62807311e-14, 3.68110488e-14, 3.70835305e-14,
                        3.74500492e-14, 3.78806206e-14, 3.81882241e-14, 3.83598383e-14,
                        3.85129976e-14, 3.87160589e-14, 3.88771617e-14, 3.88783841e-14,
                        3.87287975e-14, 3.85790047e-14, 3.85629488e-14, 3.86513683e-14,
                        3.87472621e-14, 3.87557693e-14, 3.86599871e-14, 3.85369644e-14,
                        3.84654869e-14, 3.84613117e-14, 3.84887995e-14, 3.85128523e-14,
                        3.85175253e-14, 3.84973594e-14, 3.84596280e-14, 3.84641591e-14,
                        3.85401563e-14, 3.79828905e-14, 3.60057115e-14, 3.37840954e-14,
                        3.11184881e-14, 2.70537249e-14, 2.37257664e-14, 2.07702847e-14,
                        1.71484529e-14, 1.46377481e-14, 1.14958529e-14, 9.04990219e-15,
                        7.48132714e-15, 6.46413017e-15, 5.69812871e-15, 4.93778214e-15,
                        4.15623721e-15, 3.38264766e-15, 2.65770707e-15, 2.03867656e-15,
                        1.57831690e-15, 1.27572580e-15, 1.10088734e-15, 1.02349706e-15,
                        1.01428445e-15, 1.05202436e-15, 1.11926194e-15, 1.19856389e-15,
                        1.27249695e-15, 1.32362784e-15, 1.33452328e-15, 1.28775000e-15]
                  },
                  
 I used default dfn model, and I keep getting the error that interpolation bounds were exceeded (lower bound)
 
The lowest stoichiometry that I have given is already 0.014, and the min stoichiometry in my model is 0.22

Could you please help me on this? thanks

### Steps to Reproduce

Use the above diffusivity to replace the example json file

rtimms, 2023-05-16:
Hi @chenzhao-py can you share the script and BPX file and I can take a look at resolving this? Thanks.

rtimms, 2023-05-16:
OK, I see the issue. All data is [assumed to be a function of time](https://github.com/pybamm-team/PyBaMM/blob/634f7c7389be8626c95cb22350b4e444274cceb1/pybamm/parameters/bpx.py#L352). Will fix it to handle data properly on a case-by-case basis.

chenzhao-py, 2023-05-18:
@rtimms Hey, sorry just saw your message now. Thanks for your help

ejfdickinson, 2023-05-25:
Possible short-term hack to work around: this checks through the `parameter_values` object and replace all interpolants with functions that call the interpolant with a single argument.

```
parameter_values = pybamm.ParameterValues.create_from_bpx(file)

# WORKAROUND
import copy

def make_func(data):
    def func(x):
        return pybamm.Interpolant(data["x"], data["y"], x)
    
    return func

for param,val in parameter_values.items():
    if isinstance(val, pybamm.Interpolant):
        parameter_values.update({
            param: make_func({
                    "x": copy.deepcopy(val.x[0]),
                    "y": copy.deepcopy(val.y)
            })
        })
# END
```

rtimms, 2023-05-25:
I have a PR open that fixes this but linked the wrong issue https://github.com/pybamm-team/PyBaMM/pull/2957

rtimms, 2023-05-26:
@all-contributors please add @chenzhao-py for bug

PR #2957 (2023-05-26): #2860 fix bpx data interpolant
# Description
Fixes a bug where all data included in a BPX was assumed to be given as a function of time.

Fixes #2914 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an is [... cut]
closed 2023-05-26 by rtimms