pybamm-team/PyBaMM #2939: base model

mooncode30, 2023-05-10:
output:
from .base_lithium_ion_model import BaseModel
ImportError: attempted relative import with no known parent package

why is base lithium not in the package?

tinosulzer, 2023-05-10:
You shouldn't be running the source files directly. You can run scripts or jupyter notebooks. See our examples https://github.com/pybamm-team/PyBaMM/tree/develop/examples

tinosulzer, 2023-05-10:
You have an extra space before "intercalation".

closed 2023-05-10 by tinosulzer

mooncode30, 2023-05-11:
how to estimate lithium ion soc both on DFN, SPM, SPMe models?

mooncode30, 2023-05-14:
the code input below :
submodel_options = {
" intercalation kinetics" : " asymmetric Butler - Volmer" ,
" loss of active material" : " stress and reaction- driven" ,
" SEI" : " solvent - diffusion limited" ,
" SEI film resistance" : " distributed" ,
" SEI porosity change" : " true" ,
" particle mechanics" : " swelling and cracking" ,
" lithium plating" : " irreversible" ,
" thermal" : " lumped"
}

output:
raise pybamm.OptionError(
pybamm.expression_tree.exceptions.OptionError: Option ' intercalation kinetics' not recognised. Best matches are ['intercalation kinetics', 'interface utilisation']

sorry sir, i dont understand what is the meaning of an extra space before "intercalation".?

mooncode30, 2023-05-17:
@tinosulzer hello sir, can you help me about this problem?

brosaplanella, 2023-05-17:
It should be
```
"intercalation kinetics" : " asymmetric Butler - Volmer" ,
```
instead of 
```
" intercalation kinetics" : " asymmetric Butler - Volmer" ,
```
(note there is a blank space just before intercalation which causes things to break). Also note that you have a blank space in each of the other keys of the dictionary, so you will have to fix them all.

mooncode30, 2023-05-17:
omg i got it.
thx u so much sir🙏🏻

mooncode30, 2023-05-17:
File "c:\Users\user\lion\list3.py", line 20, in <module>
    custom_params_set ["positive electrode"] = "nmc_Chen2020"
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'str' object does not support item assignment 

@brosaplanella sir, can u help me about this?