pybamm-team/PyBaMM #2954: Track down and exclude non-deterministic benchmarks

Saransh-cpp, 2023-05-15:
Currently, several benchmarks pass or fail randomly on different runs. These benchmarks should be tracked and excluded to ensure the credibility of the remaining benchmarking suite.

Note: random regressions are okay because GH Actions can be a bit noisy sometimes, but solvers failing in a particular benchmark randomly should not be okay.

jsbrittain, 2023-05-16:
This sounds like a similar issue identified for integration tests due to random initial conditions, which was resolved by wrapping the `unittest` class to fix the random seeds across all tests; see #2844

rtimms, 2023-05-16:
These non-deterministic issues worry me. It comes from [perturbing the initial conditions in the casadi solver](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/solvers/casadi_solver.py#L67), so we know why it happens, but I think for the user it is very unexpected behaviour. My vote would be to make this option `False` by default and suggest that users try setting it to `True` if the model struggles to get started. @tinosulzer, in your experience, how much difference did it make to the robustness of the solver?

tinosulzer, 2023-05-16:
It was really useful for avoiding errors at t=0 (`IDA_INIT` or something like that). But that was with casadi 3.5, so 3.6 might be more robust

Saransh-cpp, 2023-06-09:
I haven't been getting any time to work on this. I'll unassign myself and close the PR (which I don't think has something meaningful as of now). I'll pick it up again in a couple of weeks if it is still open by then.

martinjrobins, 2023-06-12:
suggestion from @brosaplanella: Looking at asv.conf.json, it seems we can specify different regression thresholds for various benchmarks (final lines of the json file): 

// "regressions_thresholds": {
  //    "some_benchmark": 0.01,     // Threshold of 1%
  //    "another_benchmark": 0.5,   // Threshold of 50%
  // },

Might be a way around it: tight thresholds for "unit" benchmarks, looser for "integration" benchmarks

brosaplanella, 2023-06-12:
> note that this particular test takes _waaaaay_ longer than the rest (when it succeeds)
> 
> ```
> False       ORegan2022    pybamm.solvers.casadi_solver.CasadiSolver   38.4±0.3s 
> ```

This is a quite tricky parameter set as it has the diffusion coefficient is very nonlinear. I would be happy to skip it the Casadi benchmark for it, or skip it altogether for the time being.

martinjrobins, 2023-06-12:
> > note that this particular test takes _waaaaay_ longer than the rest (when it succeeds)
> > ```
> > False       ORegan2022    pybamm.solvers.casadi_solver.CasadiSolver   38.4±0.3s 
> > ```
> 
> This is a quite tricky parameter set as it has the diffusion coefficient is very nonlinear. I would be happy to skip it the Casadi benchmark for it, or skip it altogether for the time being.
> 
> Also because it is likely the cause of why benchmarks take so long to solve...

I notice we're already skipping it for solve_first=true. We are also already using 30 particle grid points, so I'll turn it off fully for now

martinjrobins, 2023-06-13:
This benchmark is also occasionally timing out:

```
    GITT      Marquis2019   pybamm.models.full_battery_models.lithium_ion.dfn.DFN   pybamm.solvers.idaklu_solver.IDAKLUSolver   23.7±0.01s 
```

We do not do the corresponding benchmark for casadi (this has previously been turned off). Should we do the same for idaklu?

```
[ 95.00%] ··· time_sims_experiments.TimeSimulation.time_solve                 ok
[ 95.00%] ··· ============ ============= ======================================================= =========================================== ============
               experiment    parameter                         model_class                                       solver_class                            
              ------------ ------------- ------------------------------------------------------- ------------------------------------------- ------------
                  CCCV      Marquis2019   pybamm.models.full_battery_models.lithium_ion.spm.SPM   pybamm.solvers.casadi_solver.CasadiSolver   6.16±0.01s 
                  CCCV      Marquis2019   pybamm.models.full_battery_models.lithium_ion.spm.SPM   pybamm.solvers.idaklu_solver.IDAKLUSolver    6.19±0s   
                  CCCV      Marquis2019   pybamm.models.full_battery_models.lithium_ion.dfn.DFN   pybamm.solvers.casadi_solver.CasadiSolver    8.84±0s   
                  CCCV      Marquis2019   pybamm.models.full_battery_models.lithium_ion.dfn.DFN   pybamm.solvers.idaklu_solver.IDAKLUSolver   8.34±0.01s 
                  CCCV        Chen2020    pybamm.models.full_battery_models.lithium_ion.spm.SPM   pybamm.solvers.casadi_solver.CasadiSolver    2.77±0s   
                  CCCV        Chen2020    pybamm.models.full_battery_models.lithium_ion.spm.SPM   pybamm.solvers.idaklu_solver.IDAKLUSolver    2.78±0s   
                  CCCV        Chen2020    pybamm.models.full_battery_models.lithium_ion.dfn.DFN   pybamm.solvers.casadi_solver.CasadiSolver    5.54±0s   
                  CCCV        Chen2020    pybamm.models.full_battery_models.lithium_ion.dfn.DFN   pybamm.solvers.idaklu_solver.IDAKLUSolver    4.66±0s   
                  GITT      Marquis2019   pybamm.models.full_battery_models.lithium_ion.spm.SPM   pybamm.solvers.casadi_solver.CasadiSolver   20.2±0.01s 
                  GITT      Marquis2019   pybamm.models.full_battery_models.lithium_ion.spm.SPM   pybamm.solvers.idaklu_solver.IDAKLUSolver    20.4±0s   
                  GITT      Marquis2019   pybamm.models.full_battery_models.lithium_ion.dfn.DFN   pybamm.solvers.casadi_solver.CasadiSolver      n/a     
                  GITT      Marquis2019   pybamm.models.full_battery_models.lithium_ion.dfn.DFN   pybamm.solvers.idaklu_solver.IDAKLUSolver   23.7±0.01s 
                  GITT        Chen2020    pybamm.models.full_battery_models.lithium_ion.spm.SPM   pybamm.solvers.casadi_solver.CasadiSolver   7.59±0.01s 
                  GITT        Chen2020    pybamm.models.full_battery_models.lithium_ion.spm.SPM   pybamm.solvers.idaklu_solver.IDAKLUSolver    7.73±0s   
                  GITT        Chen2020    pybamm.models.full_battery_models.lithium_ion.dfn.DFN   pybamm.solvers.casadi_solver.CasadiSolver    12.9±0s   
                  GITT        Chen2020    pybamm.models.full_battery_models.lithium_ion.dfn.DFN   pybamm.solvers.idaklu_solver.IDAKLUSolver   11.0±0.01s 
              ============ ============= ======================================================= =========================================== ============
```

martinjrobins, 2023-06-13:
or I could increase the timeout if this benchmark is important

brosaplanella, 2023-06-13:
If it is GITT, maybe we can reduce the number of pulses tested. We currently do
```python3
"GITT": [("Discharge at C/20 for 1 hour", "Rest for 1 hour")] * 20,
```
so maybe we could do 10 pulses instead?

Saransh-cpp, 2023-06-13:
(Some of the non-deterministic benchmarks were excluded in #2784.)

PR #3031 (2023-07-06): #2954 turn off benchmark: ORegan2022 DFN solver with the casadi model
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #2954

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] Bug fix (non-breaking change which fixes an issue) [... cut]
closed 2023-07-06 by martinjrobins