pybamm-team/PyBaMM #2962: `pybamm_install_odes` is not tested anywhere

Saransh-cpp, 2023-05-16:
The script gets outdated very often because it is not being tested anywhere. The script can be tested nightly on Ubuntu once the infrastructure proposed in https://github.com/pybamm-team/PyBaMM/pull/2867 is merged.

arjxn-py, 2023-05-17:
We want to test `pybamm_install_odes` i.e. run tests associated with `scikit.odes` to avoid it from getting outdated, i believe `jax` & `jaxlib` could also be involved in here. Currently If i would want to test these, i'd have to make changes to the existing infrastructure but we want to test these in the proposed infrastructure (#2867), should I wait for it to be merged or any suggestion @Saransh-cpp, Thanks.

Saransh-cpp, 2023-05-17:
Maybe not even run tests, just check if `scikits.odes` is getting installed, as the tests are already running using `tox`. `jax` related things are already tested. We don't want to slow down the CI so we can create a new step just for testing `pybamm_install_odes` in the nightly run. I think yes, you should wait for it to be merged.

Saransh-cpp, 2023-05-19:
We can add a minimal test for this now

arjxn-py, 2023-05-19:
> We can add a minimal test for this now

Sure, on it.

closed 2023-06-12 by tinosulzer